echostack-org/echopype #3: AZFP Unpack Improvement

bnwkeys, 2018-08-27:
- [ ] (major) unpacked AZFP data is not being saved
  * will require debugging script to interrupt before plotting routine and write out data
  * unpacked AZFP data needs to be packed into netCDF format
- [ ] (major) refactor code, get rid of object-oriented paradigm
  * this is literally MacGyvered together from object-oriented code 
  * to properly create a standalone utility:
    * break script during file read routines
    * extract necessary config information, metadata, and parsing instructions
    * replicate only parsing routine in a new script and copy in static configs and meta
- [ ] (major) rev-engineer matlab scripts to check for compatibility
  * test current script against other sample data
  * check for flexibility of code to handle all use cases reading all AZFP data
    * make sensor/config agnostic
    * validate against vendor code (matlab)

leewujung, 2019-03-23:
@ngkavin welcome to the team! Let's talk more about the above soon. :)

leewujung, 2019-07-04:
Closing this with #48 and #49. :)

closed 2019-07-04 by leewujung