echostack-org/echopype #11: Combine functions from PyEcholab

erinann, 2018-08-28:
PyEchoLab and echopype were based off the same code (EchoLab in Matlab), so we want to merge them along with the changes to make use of xarray and pandas.

Tasks:

- [ ] Identify functions that are missing in either of them
- [ ] Decided which functions/methods to keep for unpacking
- [ ] Decided which functions/methods to keep for data manipulation and visualization
- [ ] Change function as in #10 to use xarray/pandas as the underlying packages instead of numpy.

erinann, 2018-08-28:
This is issue number 11 referencing old number 10 from OWH18_echopype. Yes, the old issue numbers match the new issue numbers in this case.

leewujung, 2019-07-13:
The low-level data parsing classes from pyEcholab were absorbed into echopype in #50. Right now echopype only saves the `RAW` and `NME` datagrams and ignore `DEP` and `BOT`. I think these are something to be visited later since the acoustic info is in `RAW` and lat/lon info in `NME`. It's not obviously what we gain by saving the parsed info from the other datagrams. There has to be bottom detection functions anyway. 

For the unchecked box regarding manipulation and visualization, I see them as echopype functions based on the netCDF data model and let's deal with them in the model/analysis/viz modules.

closed 2019-07-13 by leewujung