echostack-org/echopype #7: Transfer issues over from OWH/echopype repo

erinann, 2018-08-28:
When echopype repo was transferred from OWH2018 the issues did not come with it.

bnwkeys, 2018-08-28:
I don't know of any easy way to migrate issues.  From googling around I don't think the github API has that feature explicitly.  Easiest might be manual copy/paste.  There is also [this python tool](https://github.com/IQAndreas/github-issues-import) somebody wrote to do exactly this thing, but considering how few issues we had it might be more hassle than its worth to figure it out.

erinann, 2018-08-28:
Thanks @bnwkeys. Yep, it's an issue (HA!) with github that Wu-Jung, Valentina, and I ran into over the weekend. Since there aren't that many issues, I'll do the manual copy/paste.

closed 2018-08-28 by erinann