echostack-org/echopype #9: Create netCDF checker and file naming convention

erinann, 2018-08-28:
We need to make a file checker that, during the initial file parsing (.raw or .01a to netCDF), checks to see if certain acoustic parameters change. If any one of these parameters change in the middle of a .raw or .01a file, writing to the netCDF should stop, the netCFD should be saved, and a new netCDF should be opened and written to.

- Power
- Pulse duration
- Sample unit (ping rate)
- Receiver bandwidth
- Sound speed
- Start/Stop of a file

FILE NAMING. From The SONAR-netCDF4 convention for sonar data, Version 1.0.
"SONAR-netCDF4 files should always end with a “.nc” suffix to indicate that they are
a netCDF file. It is recommended that the filename should sort alphanumerically into
chronological order (e.g. date and time of the first ping in the file; thus: YYYYMMDDHHMMSS.
nc). This facilitates file management and use in analysis systems."

leewujung, 2019-01-11:
Though we don't need the checker for converting files (since the files will be converted on a 1-1 basis, see last comment), I think we still need a checker for use in calibration, processing and visualizing data. So let's keep this issue open.

leewujung, 2019-07-13:
So, change of parameters is indeed a problem as discussed in #24. I'm closing this and move the discussion. Will add the params you listed there too.

closed 2019-07-13 by leewujung