echostack-org/echopype #34: Install requires doesn't actually do anything.

lsetiawan, 2019-02-20:
## Overview

There's an issue with install requires, which is not really listing the dependencies for this package. So when performing pip install, it assumes you have all the dependencies already on your python environment. Otherwise, I can't use the package right away!

https://github.com/OSOceanAcoustics/echopype/blob/e8b735c8ef2855e434f397ccba3b1b3f5babaa4b/setup.py#L29

closed 2019-02-20 by leewujung

lsetiawan, 2019-02-20:
To show errors:
``` bash
(echopype) lsetiawan@OOI-XPS-15-9560:~/GitRepos/GitHub/staged-recipes/recipes/echopype$ pip install echopype                                                                                                [2/129]
Collecting echopype                                                                                                                                                                                                
  Downloading https://files.pythonhosted.org/packages/ae/a8/bf852901e59e118df7bbdb80b156bd319ec33469b091c2fa7b6fc5fa3fbb/echopype-0.1.2.tar.gz (10.0MB)                                                            
    100% |████████████████████████████████| 10.0MB 3.2MB/s                                               
Collecting click (from echopype)                    
  Downloading https://files.pythonhosted.org/packages/fa/37/45185cb5abbc30d7257104c434fe0b07e5a195a6847506c074527aa599ec/Click-7.0-py2.py3-none-any.whl (81kB)                                                     
    100% |████████████████████████████████| 81kB 16.1MB/s                                                
Building wheels for collected packages: echopype    
  Building wheel for echopype (setup.py) ... done   
  Stored in directory: /home/lsetiawan/.cache/pip/wheels/f6/2d/25/353756b39579f93261dfec2166931c867e69934f8add4b7564                                                                                               
Successfully built echopype                         
Installing collected packages: click, echopype      
Successfully installed click-7.0 echopype-0.1.2     
(echopype) lsetiawan@OOI-XPS-15-9560:~/GitRepos/GitHub/staged-recipes/recipes/echopype$ python           
Python 3.6.7 | packaged by conda-forge | (default, Feb 20 2019, 02:51:38)                                
[GCC 7.3.0] on linux                                
Type "help", "copyright", "credits" or "license" for more information.                                   
>>> import echopype
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/lsetiawan/miniconda/envs/echopype/lib/python3.6/site-packages/echopype/__init__.py", line 3, in <module>
    from . import convert
  File "/home/lsetiawan/miniconda/envs/echopype/lib/python3.6/site-packages/echopype/convert/__init__.py", line 1, in <module>
    from .ek60 import ConvertEK60
  File "/home/lsetiawan/miniconda/envs/echopype/lib/python3.6/site-packages/echopype/convert/ek60.py", line 18, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'

```

leewujung, 2019-02-20:
Does this mean that `install_requires` overwrites whatever that's in `requires`?

Good lesson that next time I should test install on a fresh env. :)

PR #35 (2019-02-20): Fix dependencies install requires
## Overview

This PR should resolve #34, it gets a list of dependencies from `requirements.txt`