echostack-org/echopype #36: No pip wheels

lsetiawan, 2019-02-20:
## Overview

I notice that there is no pip wheel in pypi. https://pypi.org/project/echopype/#files

I suggest creating and adding pip wheel for this package for ease of install for the user through pip. 

Though once you update the package, I can go ahead and create a conda recipe, so that this package can be installed through conda, which would make it super easy! :smile: 

Plus, I'd like to use it in my yodapy package possibly so it would be good to make the install easy! :wink:

leewujung, 2019-02-20:
Cool! I'll look into this and all of these in one upload.

I'm going to do some restructuring in terms of module/class for the data model in the coming couple weeks as I add support for another echosounder system (which, alas, is used on all non-cabled OOI networks), so just a warning here that you most likely will need to change something later one.

lsetiawan, 2019-02-20:
No worries. Let me know when you make another release! I got another PR for you. :smile:

leewujung, 2019-03-04:
[New release](https://pypi.org/manage/project/echopype/release/0.1.21/) 😎
No functionality updates but I removed the redundancy with versioneer and there's now a wheel.

lsetiawan, 2019-03-04:
Yay! That's awesome. Thanks! I will take a look and create a recipe for conda-forge!!! So exciting 😄

closed 2019-03-04 by leewujung