echostack-org/echopype #59: New release

ocefpaf, 2019-08-12:
@leewujung are you planning on using this package during the OHW19? If so can e have a new release so I can package it and make it easy to install with conda?

lsetiawan, 2019-08-12:
FYI: The latest release of `yodapy` will be using this package. Currently it's installing echopype straight from the github master branch.

leewujung, 2019-08-12:
@ocefpaf: I don't think this necessarily has to be in OHW19, but am planning on new release for user testing of AZFP data unpacking in the coming couple weeks. Will really appreciate your help to make it easy to install with conda!

@lsetiawan: the master branch should run without problem to unpack OOI Cabled Array data as they are from EK60 echosounders. There are various improvements related to actual data processing and fancier visualization that will only happen in future releases. I expect them to take a lot more time to be thoroughly tested and completed.

lsetiawan, 2019-08-12:
Sounds good! :D Thanks! Here's a little fun on using your package in yodapy! I used your example :)

[image]

ocefpaf, 2019-08-13:
> @ocefpaf: I don't think this necessarily has to be in OHW19, but am planning on new release for user testing of AZFP data unpacking in the coming couple weeks. Will really appreciate your help to make it easy to install with conda!

I got the impression from yesterday's meeting that people want to use `yodapy` and `echopype` is a dependency of `yodapy`, right?

lsetiawan, 2019-08-13:
@ocefpaf Yes, it is now a dependency of yodapy.

leewujung, 2019-08-13:
I think it is still under discussion at the moment but we should know in a
couple days. I am not sure if I'll have enough time to finish the new
release before OHW19. I don't think sonar data is critical though for the
majority of yodapy function. Also there are still processing details to be
worked out even for the MVBS plotting -- this part will not be in the new
release.

lsetiawan, 2019-08-13:
Do you think version `0.1.21` is good enough for now to at least people can play around with sonar data? https://anaconda.org/conda-forge/echopype

This is already in conda.

leewujung, 2019-08-14:
It unpacks data and performs basic calibration with default parameters, so it depends on what users want to do. 

I think conda install errors out last time I tried, but haven't been able to follow up. Something to be fixed with new release.

ocefpaf, 2019-08-15:
> I think conda install errors out last time I tried, but haven't been able to follow up. Something to be fixed with new release.

@leewujung if tried it again please report any errors in https://github.com/conda-forge/echopype-feedstock/issues and we'll take care of it.

leewujung, 2019-10-21:
@ocefpaf 0.1.21 did work, but now I am not sure how to make the new release 0.2.0 work in the feedstock... help!

ocefpaf, 2019-10-23:
See https://github.com/conda-forge/echopype-feedstock/pull/1#issuecomment-545548432

I'm working on those.

leewujung, 2020-04-24:
@ocefpaf : Just wanted to ping you here so it gets into your queue. Thanks in advance!  :) 

https://github.com/conda-forge/echopype-feedstock/pull/4 The feedstock complains about `no module named zarr` ([here](https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=147874&view=logs&j=1bf226d3-0e2f-52d8-fa93-7d9e633347b3&t=ec8d466d-e3b4-5115-4f06-222398f91e6c&l=754)) for the most recent release 0.3.1, but I am not sure where to specify that outside of requirements.txt and environment.yml   

Any guidance will be really helpful!!

leewujung, 2020-04-25:
Thought it's better to start a new issue for the latest release 0.3.1, hence closing this now.

closed 2020-04-25 by leewujung