echostack-org/echopype #66: Revise command line tool using uniform interface

leewujung, 2019-10-17:
Need to update the command line tool now that the convert and model modules have uniform interface for AZFP and EK60 data (#53). 

Specifically, need to change the input optional parameter to accommodate the additional `.XML` needed for AZFP unpacking.

prarobinson, 2019-12-02:
Currently ConvertAZFP expects two arguments: a data file and an xml file. Is the last warning above, “The XML file is ignored as it is not needed for converting", meant to be issued when the user specifies ek60, or is the goal to make xml files optional for azfp data?

leewujung, 2019-12-03:
Ouch missed the second half of the sentence. It is for if user issued an XML file for unpacking EK60 data.

closed 2019-12-03 by leewujung