echostack-org/echopype #68: Whitespace in path breaks EK60 conversions

adhameed, 2019-10-26:
Example path: "/home/user/source/oops i spaced again/filename.raw"

Code to reproduce: very first file conversion example in the docs
```python
from echopype.convert import Convert
data_tmp = Convert('/a/path/that has spaces/FILENAME.raw')
data_tmp.raw2nc()
```

Traceback after running my test (and crashing):
```
  File "/Users/user/source/oops i spaced again/test.py", line 7, in <module>
    ek60_converter.raw2nc()
  File "/Users/user/source/echopype/venv/lib/python3.7/site-packages/echopype/convert/ek60.py", line 888, in raw2nc
    grp.set_toplevel(_set_toplevel_dict())  # top-level group
  File "/Users/user/source/echopype/venv/lib/python3.7/site-packages/echopype/convert/ek60.py", line 695, in _set_toplevel_dict
    out_dict['date_created'] = dt.strptime(fm.group('date') + '-' + fm.group('time'),
AttributeError: 'NoneType' object has no attribute 'group'
```

Tested on (and failing on) Mac OS 10.14.6 and Windows 10 Pro 1903.

Solution: if I remove the spaces in my test paths, files are parsed correctly.

leewujung, 2019-10-28:
Thanks for reporting the bug. Will get this fixed soon!

leewujung, 2019-11-19:
@prarobinson: could you also take a look at this? Something is not quite right with parsing with filepath passed into the object. Thx!

Quick background info: most EK60 files are named with the convention `[SURVEY]-D[YYYYMMDD]-T[HHMMSS].raw`.

prarobinson, 2019-12-02:
Fixed this. I used a non-re string parsing solution.

closed 2019-12-03 by leewujung