echostack-org/echopype #78: Basic echogram plots

leewujung, 2019-12-10:
We'll add a thin wrapper to the xarray plotting function to plot basic echograms (Sv or MVBS).

- x-axis:
  - actual time stamp (coordinate `ping_time`) or
  - ping number (just plain numbers 1-n).
- y-axis: three options here:
  - range in meters
  - depth in meters (corrected for tilt if known)
  - time in milliseconds
- Colormap: colormap is an issue here. The fisheries acoustics community is used the so called EK500 colormap (below), and Matlab users are very much used to jet. We should promote colorblind-friendly colormaps. Magma seems to work well among the matplotlib choices.

- EK500 colormap RGB values, from [IMOS BASOOP](http://imos.org.au/fileadmin/user_upload/shared/SOOP/BASOOP/viz_sv_html/viz_sv.html#24). We should make this an option for plotting echogram.
  ```matlab
  function [EK500cmap]=EK500colourmap()
    % EK500colourmap is the colour map used by EK500
        EK500cmap = [255   255   255   % white
            159   159   159   % light grey
            95    95    95   % grey
            0     0   255   % dark blue
            0     0   127   % blue
            0   191     0   % green
            0   127     0   % dark green
            255   255     0   % yellow
            255   127     0   % orange
            255     0   191   % pink
            255     0     0   % red
            166    83    60   % light brown
            120    60    40]./255;  % dark brown
    end
  ```

- We'll need to decide when the echogram becomes too large to be plotted directly using matplotlib and need to use tools like datashader. Need to experiment with loading/plotting many files. With the reduced size of MVBS it's likely not an issue, but for plotting the calibrated Sv or raw data, it can be slow depending on the ping rate and data time span.

lsetiawan, 2021-05-24:
Just found this issue, was thinking of creating this by getting values from pictures online haha. Here's a python version of the above function (**I've divided the above values by 255**):

```python
from matplotlib.colors import ListedColormap

cm_data = [[1.0, 1.0, 1.0],
 [0.6235294117647059, 0.6235294117647059, 0.6235294117647059],
 [0.37254901960784315, 0.37254901960784315, 0.37254901960784315],
 [0.0, 0.0, 1.0],
 [0.0, 0.0, 0.4980392156862745],
 [0.0, 0.7490196078431373, 0.0],
 [0.0, 0.4980392156862745, 0.0],
 [1.0, 1.0, 0.0],
 [1.0, 0.4980392156862745, 0.0],
 [1.0, 0.0, 0.7490196078431373],
 [1.0, 0.0, 0.0],
 [0.6509803921568628, 0.3254901960784314, 0.23529411764705882],
 [0.47058823529411764, 0.23529411764705882, 0.1568627450980392]]

ek500_cmap = ListedColormap(cm_data)
```

I tried it and looks great! 😄 

[image]

leewujung, 2021-05-24:
@lsetiawan the colormap looks great, just a detail that requires a small tweak:
The first and the last color code should be treated as above or below the vmax and vmin, like below:
[image]
I think it'll be better to use the RGB values and explicitly divide 255 so that it can be easily reproduced.

We also need to fix the vertical axis via a couple helper functions. :)

lsetiawan, 2021-08-12:
Notes for desired plots:
1. If user passes in EchoData, plot the `backscatter_r` in beam group.
2. If user passes `MVBS` / `Sv`, then plot those directly.
3. Add a custom EK500 colormap along with the cmocean colormaps.

lsetiawan, 2021-08-12:
Note for water_level parameter in platform group for depth_offset. See #259 for related issue.

leewujung, 2021-12-19:
Forgot that there was this one that can now be closed! :D

closed 2022-02-27 by leewujung