echostack-org/echopype #135: Change naming of the model package

leewujung, 2020-05-02:
The `model` package performs calibration and depth bin+time bin-averages to produce MVBS, plus noise removal that has to be done before the averaging. However `model` is not a good name and is remnant of the early days when I was trying to use the model-view-controller design pattern. But this name is easily confused with physics-based scattering models.

Given what it does and that more analysis functions will be implemented in the `analyze` package, I think we can change `model` to `process`.

Along with the above discussion, @emiliom also suggested to simplify the importing to allow just `import echopype.EchoData`. With the name change, it should probably become `import echopype.Process`. The same could be done for the `convert` package so can do `import echopype.Convert as Convert` rather than `from echopype.convert import Convert`.

emiliom, 2020-05-03:
To be clearer, using `Convert` as an example, what I'm thinking of is adding:
```python
from .convert import Convert
```
to [echopype/__init__.py](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/__init__.py).

Though I haven't tested any of this, I think that will enable these sort of less verbose access: 

```python
# 1
import ecopype
data = echopype.Convert('inputargs')

# 2
from echopype import Convert
data = Convert('inputargs')
```

emiliom, 2020-05-11:
BTW, when you get around to renaming the `model` subpackage to `process`, make sure to do it using `git mv`, not the file system `mv`. That way the git history is preserved, otherwise you lose the commit history for each module.

leewujung, 2020-05-13:
A note for when I get around to do this: 
`from .convert import Convert` was done in earlier version but then removed because I couldn't make it work with `sphinx-automodapi` to auto-generate API refs. Now that this sphinx extension seems to throw off readthedocs (remain to be fixed on PR #141 ) maybe it's time to switch to something else.

leewujung, 2020-06-21:
This is closed by #156.

PR #156 (2020-06-13): Rename Model to Process
Also renames EchoData to Process.

closes #135
PR #160 (2020-06-19): Change echopype.model to echopype.Process
This PR contains changes that rename `echopype.model` to `echopype.Process` as well as the class `EchoData` to `Process` for consistency. The documentation and notebook are also changed.

The `analysis` subpackage is removed for now since it's currently empty. We'll add it back when there are some functions, probably in the name of `analyze` so that all subpackages names are verbs.

This closes #135, closes #158 , and also closes #149.
closed 2020-06-21 by leewujung