echostack-org/echopype #148: raw2zarr, raw2nc, save: rename, consolidate?

emiliom, 2020-05-27:
I was originally going to open this issue as a suggestion to rename `raw2zarr` and `raw2nc` to something like `to_zarr` and `to_netcdf`. I find the `raw2...` names somewhat confusing and misleading. In reality, the export (serialization) is not actually *from* a raw format, but rather from an in-memory xarray dataset, right? In other words, the raw data has already been read and mapped to the data convention in xarray. The names I'm suggesting match xarray export functions. 

But looking at [the code](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/convert/convertbase.py#L129), I see that at this time there's no reason to have two separate export functions `raw2zarr` and `raw2nc`. They're effectively identical, except for the `param` passed to `self.save`. 

I also noticed that `raw2zarr` currently seems to be limited to writing to a file system path and doesn't seem to accommodate writing to object storage like S3 (which xarray's `to_zarr` supports). If that's correct, that's a separate issue. Adding that functionality would be a good reason to maintain two separate functions, as xarray does.

leewujung, 2020-05-27:
I think changing `raw2zarr` and `raw2nc` to `to_zarr` and `to_netcdf` is a great suggestion. This way it is consistent with xarray and pandas methods naming so familiar to users.

In terms of having separate methods, the conversion to zarr and netcdf needs to be separate right now because the way to combine data from multiple raw files into a single zarr or nc file is different for zarr and netcdf under xarray (`to_netcdf` does not allow appending variables directly). @ngkavin : could you finalize the block diagram we went through together last time and upload it to under /docs/img so that we can discuss further based on that? Thanks!

For the `raw2zarr` method to accommodate writing to an object storage, under the hood it is using xarray's `to_zarr` method, so the `validate_path` method is probably where the tweaking needs to go. @emiliom : could you raise a separate issue on this? Thanks!

emiliom, 2020-05-27:
> I  think changing raw2zarr and raw2nc to to_zarr and to_netcdf is a great suggestion. This way it is consistent with xarray and pandas methods naming so familiar to users.

Thinking through things a bit further, I wonder if reusing the names from xarray could cause confusion, if the parameters are actually different ... I'll look into it further, but not right away.

Still, my point about the `raw2` part stands: it's misleading to say the export (serialization) is happening from the raw format per se. Also, raw2zarr and raw2nc apply the class method `save`; they're practically aliases for save. From a developer perspective, it's a bit disorienting (harder to learn) to have this functionality be referred to as "save" in one context and "raw2\<output\>" in another

ngkavin, 2020-05-28:
[This](https://drive.google.com/open?id=1yrlQQXKQgcfZcTh9EZu-AJrqCamFrRPL) is a link to the diagram that shows the flow of saving a file to NetCDF or Zarr I used for reference.

emiliom, 2020-05-28:
Nice, thanks!

leewujung, 2021-04-25:
This issue can be closed one `class-redesign` is merged to `master`.

leewujung, 2021-04-27:
Closing this with #298.

closed 2021-04-27 by leewujung