echostack-org/echopype #149: Remove analysis subpackage, for now

emiliom, 2020-05-27:
The [analysis](https://github.com/OSOceanAcoustics/echopype/tree/convert-refactor/echopype/analysis) subpackage is just an empty placeholder at this time, at least in both master and the convert-refactor branches. It hasn't been touched in 16 months. Given that there's already plenty of changes in place for a new release, it seems unlikely (and unwise?) that you'll be fleshing out `analysis` for this release. To a user, I think it's confusing and misleading to have it there. I suggest removing it until you're actually ready to add that functionality.

leewujung, 2020-05-27:
This sounds good. I'll remove it when merging the `convert-refactor` branch for the new release.

leewujung, 2020-06-21:
This is closed by #160.

PR #160 (2020-06-19): Change echopype.model to echopype.Process
This PR contains changes that rename `echopype.model` to `echopype.Process` as well as the class `EchoData` to `Process` for consistency. The documentation and notebook are also changed.

The `analysis` subpackage is removed for now since it's currently empty. We'll add it back when there are some functions, probably in the name of `analyze` so that all subpackages names are verbs.

This closes #135, closes #158 , and also closes #149.
closed 2020-06-21 by leewujung