echostack-org/echopype #150: raw2zarr support for object store reference

emiliom, 2020-05-27:
`raw2zarr` currently seems to be limited to writing to a file system path and doesn't seem to accommodate writing to object storage like S3 (which xarray's `to_zarr` supports).

For the raw2zarr method to accommodate writing to an object storage, under the hood it is using xarray's to_zarr method, so the `validate_path` method is probably where the tweaking needs to go.

emiliom, 2020-09-25:
Is this capability now implemented? I have a hunch it is, but it'd be good to get confirmation.

ngkavin, 2020-09-25:
`to_zarr` currently only supports writing to a local file path, but seeing as xarray  already supports saving to object storage, it will probably get added soon.

emiliom, 2020-12-24:
*Copying over @ngkavin's Oct 28 comment from https://github.com/OSOceanAcoustics/echopype/issues/195#issuecomment-718059751:*

I implemented a way to save to a cloud store simply by requiring a `MutableMapping` to a cloud store and passing that mapping into `xarray`/`zarr`. You can see how it would be used here: https://github.com/ngkavin/echopype/blob/class-redesign/echopype/tests/test_convert_ek60.py#L96

It was done this way to remove dependencies on s3fs/gcfs/adlfs but echopype can still take in mappings created with these packages. Please let me know if something doesn't seem right (I just recently started learning about how the cloud works).

leewujung, 2021-04-25:
This issue can be closed one `class-redesign` is merged to `master`.

leewujung, 2021-04-27:
Closing this with #298.

closed 2021-04-27 by leewujung