echostack-org/echopype #171: Save or export EK80 XML configuration datagram

leewujung, 2020-06-24:
Currently only the fields of the EK80 XML configuration datagram that correspond to requirements in the SONAR-netCDF convention is saved into netCDF/zarr. However, there may be other parameters that will be useful for users later. Therefore it seems a good idea to give users access to the entire datagram once the data is parsed.

@ngkavin and I discussed the options and seems the following would make sense:
- export a companion XML file with netCDF/zarr, and/or
- save the entire config datagram into the Vendor group

@emiliom @lsetiawan @valentina-s any suggestions?

leewujung, 2020-06-30:
@ngkavin and I decided to move forward with:
- giving user the option to export the XML file, as a separate method -- description above updated
- test saving the entire XML string as a variable in the Vendor group: see if it is allowed in netCDF and zarr, and if the file size increases dramatically

emiliom, 2020-06-30:
Is there *data* in the XML file, or is it just configuration information? If there's no actual data, it seems unlikely including it in the netcdf / zarr would increase file size dramatically. 

Is the EK80 XML file fairly flat, or does it have deep hierarchies? If the former, wouldn't it make more sense to convert from XML per se to netcdf/zarr attributes. Do you have an example EK80 XML you could point to?

leewujung, 2020-06-30:
> Is there data in the XML file, or is it just configuration information? If there's no actual data, it seems unlikely including it in the netcdf / zarr would increase file size dramatically.

It depends... (as always!). Most part of the hierarchy is just config information, but for some files (see below), some transducer calibration parameters are also present and those are more like actual data. 

The calibration parameters are currently not saved into netCDF/zarr. I don't think the original SONAR-netCDF has fields designed for them, but we can certainly create new data variables in the Vendor group (which is designed for saving anything additional that are specific to manufacturers). BUT if we were to store the XML string into a data variable, that'll get long.

> Is the EK80 XML file fairly flat, or does it have deep hierarchies? If the former, wouldn't it make more sense to convert from XML per se to netcdf/zarr attributes. Do you have an example EK80 XML you could point to?

Here are two:
1. [This one](https://drive.google.com/file/d/1-l7BWJSaELsM72Ps5xB1UOEJDvA0bEaY/view?usp=sharing) is typical of those simple files without calibration parameters.
2. [This one](https://drive.google.com/file/d/1g1M-iSm7xF3G9ydkhnvLQbeg7O-1AkWt/view?usp=sharing) is typical of those files with additional calibration parameters.

emiliom, 2020-06-30:
Thanks. How about loading those sample XML files on github? It doesn't have to be this repo; it could be a different repo in this organization, if you'd prefer. It just makes it easier to examine the XML files more quickly in more flexible ways ... Another option that's potentially easier to do but a bit more opaque is loading them as gists.

leewujung, 2020-06-30:
I think downloading the files and opening them in browser does the job. That's how I debugged the config header parsing problems. :)

PR #179 (2020-08-19): Save EK80 configuration xml
Saves the configuration xml to the vendor group.
Also provides the option for exporting the configuration to an xml file by calling `export_xml()` on a `Convert `object

Closes #171
closed 2020-08-19 by leewujung