echostack-org/echopype #173: Something seems awry with echopype 0.4.0

tsgouros, 2020-06-29:
On a fresh colab, after 'pip install echopype' and 'from echopype import convert':

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-7-fee2cb6d1ab0> in <module>()
----> 1 from echopype import convert

4 frames
/usr/local/lib/python3.6/dist-packages/echopype/convert/utils/ek_raw_io.py in <module>()
     10 import struct
     11 import logging
---> 12 import echopype.convert.utils.ek_raw_parsers as parsers
     13 
     14 __all__ = ['RawSimradFile']

AttributeError: module 'echopype' has no attribute 'convert'
```
This error happens with "from echopype import Convert" also.

  ------------------

Here's the pip install log, all seems clean for echopype-0.4.0
```
Collecting echopype
  Downloading https://files.pythonhosted.org/packages/d7/91/430472654b3a678f95a6b28d0dac58d752d5c0ea0d5a90d919e56273340c/echopype-0.4.0-py3-none-any.whl (126kB)
     |████████████████████████████████| 133kB 8.5MB/s 
Requirement already satisfied: matplotlib in /usr/local/lib/python3.6/dist-packages (from echopype) (3.2.2)
Requirement already satisfied: numpy in /usr/local/lib/python3.6/dist-packages (from echopype) (1.18.5)
Requirement already satisfied: dask[array] in /usr/local/lib/python3.6/dist-packages (from echopype) (2.12.0)
Collecting pynmea2
  Downloading https://files.pythonhosted.org/packages/88/5f/a3d09471582e710b4871e41b0b7792be836d6396a2630dee4c6ef44830e5/pynmea2-1.15.0-py3-none-any.whl
Requirement already satisfied: pytz in /usr/local/lib/python3.6/dist-packages (from echopype) (2018.9)
Requirement already satisfied: scipy in /usr/local/lib/python3.6/dist-packages (from echopype) (1.4.1)
Requirement already satisfied: xarray in /usr/local/lib/python3.6/dist-packages (from echopype) (0.15.1)
Collecting netCDF4
  Downloading https://files.pythonhosted.org/packages/35/4f/d49fe0c65dea4d2ebfdc602d3e3d2a45a172255c151f4497c43f6d94a5f6/netCDF4-1.5.3-cp36-cp36m-manylinux1_x86_64.whl (4.1MB)
     |████████████████████████████████| 4.1MB 15.3MB/s 
Collecting zarr
  Downloading https://files.pythonhosted.org/packages/a3/87/383d77399148ef0772da3472b513ecf143252e7c365c51b0f06714800366/zarr-2.4.0.tar.gz (3.3MB)
     |████████████████████████████████| 3.3MB 47.1MB/s 
Requirement already satisfied: python-dateutil>=2.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib->echopype) (2.8.1)
Requirement already satisfied: cycler>=0.10 in /usr/local/lib/python3.6/dist-packages (from matplotlib->echopype) (0.10.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib->echopype) (2.4.7)
Requirement already satisfied: kiwisolver>=1.0.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib->echopype) (1.2.0)
Requirement already satisfied: toolz>=0.7.3; extra == "array" in /usr/local/lib/python3.6/dist-packages (from dask[array]->echopype) (0.10.0)
Requirement already satisfied: setuptools>=41.2 in /usr/local/lib/python3.6/dist-packages (from xarray->echopype) (47.3.1)
Requirement already satisfied: pandas>=0.25 in /usr/local/lib/python3.6/dist-packages (from xarray->echopype) (1.0.5)
Collecting cftime
  Downloading https://files.pythonhosted.org/packages/0f/5e/ee154e2aabb0beea0c4c7dc3d93c6f64f96a2a2019bbd05afc905439d042/cftime-1.1.3-cp36-cp36m-manylinux1_x86_64.whl (322kB)
     |████████████████████████████████| 327kB 38.9MB/s 
Collecting asciitree
  Downloading https://files.pythonhosted.org/packages/2d/6a/885bc91484e1aa8f618f6f0228d76d0e67000b0fdd6090673b777e311913/asciitree-0.3.3.tar.gz
Collecting fasteners
  Downloading https://files.pythonhosted.org/packages/18/bd/55eb2d6397b9c0e263af9d091ebdb756b15756029b3cededf6461481bc63/fasteners-0.15-py2.py3-none-any.whl
Collecting numcodecs>=0.6.4
  Downloading https://files.pythonhosted.org/packages/53/2a/1dc435cbd1d082827190a3e46168fd04f74e266e91313969d5a1aab601bf/numcodecs-0.6.4.tar.gz (3.8MB)
     |████████████████████████████████| 3.8MB 53.5MB/s 
Requirement already satisfied: six>=1.5 in /usr/local/lib/python3.6/dist-packages (from python-dateutil>=2.1->matplotlib->echopype) (1.12.0)
Collecting monotonic>=0.1
  Downloading https://files.pythonhosted.org/packages/ac/aa/063eca6a416f397bd99552c534c6d11d57f58f2e94c14780f3bbf818c4cf/monotonic-1.5-py2.py3-none-any.whl
Building wheels for collected packages: zarr, asciitree, numcodecs
  Building wheel for zarr (setup.py) ... done
  Created wheel for zarr: filename=zarr-2.4.0-cp36-none-any.whl size=127066 sha256=692b004f7d7b5ac3c95c78ee12fe1d4b299e9dac1e30a26488dac99ea31cab67
  Stored in directory: /root/.cache/pip/wheels/e1/5b/25/24c685604b91139aba00a5b6299b53e7a0661f737f27782559
  Building wheel for asciitree (setup.py) ... done
  Created wheel for asciitree: filename=asciitree-0.3.3-cp36-none-any.whl size=5037 sha256=2db448a1d00ce2c005387d71d62f64324ea5b972e38c1a98bf6517d8e05d621c
  Stored in directory: /root/.cache/pip/wheels/1d/d9/58/9808b306744df0208fccc640d3d9952a5bc7468502d42897d5
  Building wheel for numcodecs (setup.py) ... done
  Created wheel for numcodecs: filename=numcodecs-0.6.4-cp36-cp36m-linux_x86_64.whl size=3879707 sha256=cc89c1c1ce8a43b860bfa8b1321803b39644472bc2a5dc995faf5be4a63c5ab6
  Stored in directory: /root/.cache/pip/wheels/ca/07/ed/fea2e120cbb91d90b577c5ac56b4b082024f56fcd88e9afa55
Successfully built zarr asciitree numcodecs
Installing collected packages: pynmea2, cftime, netCDF4, asciitree, monotonic, fasteners, numcodecs, zarr, echopype
Successfully installed asciitree-0.3.3 cftime-1.1.3 echopype-0.4.0 fasteners-0.15 monotonic-1.5 netCDF4-1.5.3 numcodecs-0.6.4 pynmea2-1.15.0 zarr-2.4.0
```

leewujung, 2020-06-29:
Hm, I don't have this problem when pip install locally. What happens when you try in your local setup?

tsgouros, 2020-06-29:
Seems to work on local machine. Behavior is different on colab if I do '!pip install -Iv echopype==0.3.2'. Won't let me do 'from echopype import Convert' but I can do 'from echopype import convert' and 'import echopype', neither of which work with 0.4.0.

leewujung, 2020-06-29:
Interesting. We did change the import statement slightly in v0.4.0 (part of #135). Let me take a look.

leewujung, 2020-06-29:
@tsgouros can you give this a try: `!pip install git+https://github.com/leewujung/echopype.git@import-fix#egg=echopype`

It works for me on colab now. If this runs fine no problem, I'll combine this with the other fix on two-in-one transducer config parsing in v0.4.1.

tsgouros, 2020-06-29:
Appears to work, thank you.

tsgouros, 2020-06-29:
Or maybe I should let you close it.

PR #174 (2020-06-29): Fix import issue on Google Colab
This PR changes all submodule import statements to relative paths, which fixes import problems in Google colab.

This closes #173 .
closed 2020-06-29 by leewujung