echostack-org/echopype #176: MVBS plots show anomalous jumps in the range_bin values

cwingard, 2020-06-29:
I've been exploring the use of echopype for the OOI AZFP data and have created a [notebook](https://nbviewer.jupyter.org/github/cwingard/ooi-data-explorations/blob/echopype/python/examples/notebooks/zplsc/CE01ISSM-00004%20Echogram%20for%202015-10-09.ipynb) showing the steps I followed based off of your example notebooks and [documentation](https://echopype.readthedocs.io/en/latest/).

When plotting the data created using the `Process.get_MVBS` function, I'm seeing jumps in the range_bin values that do not correspond with the data. An example plot is available towards the bottom of the following [notebook](https://nbviewer.jupyter.org/github/cwingard/ooi-data-explorations/blob/echopype/python/examples/notebooks/zplsc/CE01ISSM-00004%20Echogram%20for%202015-10-09.ipynb)

leewujung, 2020-06-29:
Interesting. It seems would have something to do with the rounding up for depth bins.

In cell 13, if you print out the MVBS output, do you see a row of NaNs for the "shorter" (those with the empty spot on top) averaged bins?

Since the MVBS is calculated discontinuously (which needs to be fixed and is related to #54 ), the binned range_bin could be different. Before looking into the data, I don't immediately see why this happens, unless the original (Sv) range_bin size changes over the period of time from the various files?

cwingard, 2020-06-29:
Looking at the MVBS, I do see 6 rows of NaNs for the the "shorter" bins. I looked at the individual NetCDF files created by the `get_MVBS` function and the number of elements for range_bin does change from 121 for the "shorter" to 127 for the "longer".

leewujung, 2020-06-29:
Ok, thanks for looking into these. I'll look into what's going on. It may take a few days for me to get to this, but the MVBS calculation is one of the things that definitely needs more work and it's great that this bug pushes it to higher priority!

ngkavin, 2020-07-05:
The discontinuities are due to each file having a separate temperature array which affects the sound speed which is used in calculating the tile sizes for the MVBS calculation. Calculating MVBS by averaging across all of the temperatures in the file would fix the issue.

leewujung, 2020-07-05:
@ngkavin : thanks for looking into this! 

So, in cell 8 if temperature is specified like salinity and temperature, the bin size would be identical for all files. The discrepancy in cell 13 is therefore due to differences in `MVBS_range_bin_size`, which is the size in meter for each averaged `range_bin`. 

This is interesting, because it points to:
1. a wrapper visualization that takes care of this in plotting, and 
2. an interpolation function that can resolve this type of discrepancy when merging data, if we want to use values from the closest CTD cast values in calibration. A function getting passed into `xr.open_mfdataset` by `preprocess` would do the job.

And obviously, the ongoing #177 that would fix #175 would solve this from a different route, but providing the above would be useful since it's not always the case that the data would be merged when converted.

leewujung, 2021-08-07:
Circling back to this issue after a long time. :) 

This should be resolved by using `compute_MVBS` from v0.5.0 in which MVBS is computed by binning using the actual range coordinate in meters, rather than the number of bins in a hacky way.

There will still be cases when a file has the max range at the furthest MVBS bin to be 1 bin longer or shorter than the others, due to the unlucky changes in the recorded temperature, which is used to compute the range. This can be mitigate by setting the C, T, D values through `env_params` when computing Sv, so that if the number of samples bins is the same, the max range will be the same.

leewujung, 2022-11-24:
I think this long-standing issue can be closed now for the following reasons:
- The improved `compute_MVBS` function allows varying `echo_range` parameter across ping time.
- The "unlucky" situation where small temperature variation results in an extra bin along `echo_range` is unavoidable and is something the users should make a conscious decision on -- whether they want to use a uniform temperature to avoid this extra bin, or that this extra bin is "real" and downstream analysis code would take this into account.

closed 2022-11-24 by leewujung