echostack-org/echopype #184: Uncaught KeyError creating out_dict in convert/ek80.py

Chuck-A, 2020-07-23:
After commenting out time format error from issue #182 I had the following error when parsing a Saildrone file:

  File "/Users/charles.anderson/opt/miniconda3/lib/python3.7/site-packages/echopype/convert/ek80.py", line 270, in _set_platform_dict
    out_dict['water_level'] = self.environment['water_level_draft']
KeyError: 'water_level_draft'

leewujung, 2020-07-23:
@Chuck-A I think a quick fix could be to set it to `NaN` and spit out an warning to let user know that it is empty.

Could you give that a try? Thanks!

Chuck-A, 2020-07-23:
Will do :)

-- 
Chuck Anderson
Fisheries Acoustic Data Manager
NOAA National Centers for Environmental Information (NCEI)
formerly NOAA’s National Geophysical Data Center
Cooperative Institute for Research in Environmental Sciences (CIRES)
at University of Colorado at Boulder
325 Broadway, E/NE42
Boulder, Colorado 80305-3328
Tel: 303-497-4742; Fax: 303-497-6513

leewujung, 2020-07-23:
Thanks! Just to add more info: @doron2402 said that -1.9 m was what Saildrone had.

I think this calls for a method for user to set this before conversion, so that the .nc or .zarr would have that info included.

leewujung, 2020-07-24:
This was addressed in #187 .

closed 2020-07-24 by leewujung