echostack-org/echopype #193: KeyError: 'longitude' when using raw2nc with EK60 data

PierrePriou, 2020-09-22:
I received the following error when trying to convert raw EK60 data into NetCDF using `raw2nc`

[image]

The data was collected in the Arctic and, generally, had bad GPS fixes. I am currently using Echopype version 0.4.0. 

To reproduce the error, I attached the code that I used and a [link](https://drive.google.com/file/d/1zaKpw4I2XN2g6JPYV42SvUAzjFbugNPq/view?usp=sharing) for downloading the raw data file. 

```{python raw2nc error, include = T, eval = F}
import os
os.chdir("C:/Users/cfer/PhD/Echopype_test")
os.getcwd()

file_name = "EK60_2019_LEG_02-D20190717-T095103"

from echopype import Convert
dc = Convert('data/raw/acoustics/' + file_name + '.raw') 
dc.raw2nc() # KeyError: longitude
````
Thank you.

leewujung, 2020-09-24:
@PierrePriou: thanks for reporting this. 

When you said "bad GPS fixes", do you know if the GPS was actually sending messages? If the GPS message stream was empty, this is something we had put in a fix for (see #185). I still have to do a release for that to be formal, hopefully in the coming few days.

If you could give the latest master branch a try by `pip install git+https://github.com/OSOceanAcoustics/echopype` that'll be great. If there's additional problem please kindly let us know.

cyrf0006, 2020-09-24:
@leewujung: I tested @PierrePriou 's problem with he latest echopype version. The problem remains. It looks like the NMEA message is troncated in his files:

`$INGGA,095105.067,7244.26485,N,06646 [...] `

instead of 

`$INGGA,144001.10,6759.823526,N,00517.812805,E,xxx*7C`

I think that you already figured out the necessity of correcting the problem L. 365 of [ek60.py](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/convert/ek60.py) where you say:

` # TODO: use NaN when nmea_msg is empty`

I think that testing the message if broken and replace with NaNs where needed would solve the problem. I am sorry I am too busy to correct this now.

leewujung, 2020-09-27:
@cyrf0006 : turns out that we did fix that, but on a different branch... I've added the fix as a patch (#197 ) to the master as part of a minor release along with a few other, before the other one is ready.

@PierrePriou : you should be able to parse the file now, but it appears that the file you provided have partially correct readings and a large number of empty ones and a large erroneous spike.

@ngkavin : please check [here](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/convert/ek60.py#L367-L368) for the case when only one valid NMEA message is found in the datagrams and add that to `class-redesign`. Without that it errored out for me for the test file `2019118 group2survey-D20191214-T081342.raw`.

cyrf0006, 2020-09-28:
I confirm that it works now! 
Thanks @leewujung & @ngkavin!

PierrePriou, 2020-09-28:
It is working thank you @leewujung, @ngkavin and @cyrf0006 for the help! 

Unfortunately, the GPS malfunctioned during that cruise so that may be the reason behind the lack of data.

leewujung, 2020-09-28:
@PierrePriou hopefully you had other backups to get the info! I'll close this now. :)

commit 4ace1944 (2020-12-01): nmea string fix closes #193
closed 2020-09-28 by leewujung