echostack-org/echopype #195: Enhance Convert to accept file-like objects, not just local file path strings

emiliom, 2020-09-25:
Currently `Convert` can read only files from the locally accessible file system. It accepts one or more file path strings. That's true of both the current (`master`) version and the refactored version in `class-redesign`.

If we want to read a file not on the file system or a buffered stream in memory, we're out of luck.  The only option right now is to download the file, then read it as a local file.

Having `Convert` also accept [file-like objects](https://docs.python.org/3/glossary.html#term-file-like-object) would open up this capability. It would enable both `io.ByteIO` and `io.StringIO` (see [here]() and [here]() -- references for myself, if nothing else!) as well as [fsspec](https://filesystem-spec.readthedocs.io/)-based access to remote files (S3 buckets, http-hosted files, etc).

I'm interested in attempting a limited implementation myself. For my own reference when I get back to this, I'm listing here the sequence of method calls for EK60 that currently is traversed to the statement that opens a local file:

1. [convert/convert.py#L11](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/convert/convert.py#L11)
2. [convert/ek60.py#L39](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/convert/ek60.py#L39)
3. [convert/ek60.py#L253](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/convert/ek60.py#L253)
4. [convert/utils/ek_raw_io.py#L64](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/convert/utils/ek_raw_io.py#L64)

emiliom, 2020-09-26:
An `S3FileSystem` (type:  `s3fs.core.S3FileSystem`) has an `open` method:

```python
Signature: FS.open(path, mode='rb', block_size=None, cache_options=None, **kwargs)
Docstring:
Return a file-like object from the filesystem

The resultant instance must function correctly in a context ``with``
block.

Parameters
----------
path: str
    Target file
mode: str like 'rb', 'w'
    See builtin ``open()``
block_size: int
    Some indication of buffering - this is a value in bytes
cache_options : dict, optional
    Extra arguments to pass through to the cache.
encoding, errors, newline: passed on to TextIOWrapper for text mode
```

lsetiawan, 2020-10-27:
I've started playing around with this to proceed with my usecase for CAVA. The code changes can be found on my [convert_http](https://github.com/lsetiawan/echopype/tree/convert_http) branch. Currently I'm only working on EK60, but I think this solution can be used for the others also.

1. https://github.com/lsetiawan/echopype/blob/convert_http/echopype/convert/convert.py#L53-L60
    Here I am using `urllib` and `fsspec` to check on the raw file. `fsspec` syntax is similar to `os` so `isfile` works for all cases.
2. https://github.com/lsetiawan/echopype/blob/convert_http/echopype/convert/convertbase.py#L143 and https://github.com/lsetiawan/echopype/blob/convert_http/echopype/convert/convertbase.py#L157
    I am using current directory rather than dirname since no download happened. Not sure if we want to differentiate the two.
3.  https://github.com/lsetiawan/echopype/blob/convert_http/echopype/convert/ek60.py#L244-L296
    Checks has to happen to separate the url source or local file source, so I had to create a private function with the same name `load_ek60_raw`.
4. https://github.com/lsetiawan/echopype/blob/convert_http/echopype/convert/utils/ek_raw_io.py#L92-L96
    To support the `load_ek60_raw`, the `RawSimradFile` has to be able to take an fsspec HTTPFile object. I have also updated the superclass calls in this code to be up to date with python3 superclass spec.

Those are what I got so far. My questions are for output file, should a user be able to upload to a server or object storage place?

In that same branch changes to ProcessEK60 has been made to be able to read zarr file from s3 bucket, and it works. Let me know if you'd like me to make a PR to the official echopype. In the mean time, try out the branch :smile:. Thanks!

emiliom, 2020-10-28:
Wow, this is awesome @lsetiawan! I never got around to working on this; my acoustics time (and also a lot of @leewujung's time) got sucked into a proposal instead. But I think @ngkavin was looking into this, or maybe it was the [task of writing/uploading directly to object storage](https://github.com/OSOceanAcoustics/echopype/issues/150).

Now I want to have the time to test your branch. Hopefully in a few days :pray:

ngkavin, 2020-10-28:
@lsetiawan I agree this is great! I will be sure to test it out soon.

I implemented a way to save to a cloud store simply by requiring a `MutableMapping` to a cloud store and passing that mapping into `xarray`/`zarr`. You can see how it would be used here: https://github.com/ngkavin/echopype/blob/class-redesign/echopype/tests/test_convert_ek60.py#L96

It was done this way to remove dependencies on s3fs/gcfs/adlfs but echopype can still take in mappings created with these packages. Please let me know if something doesn't seem right (I just recently started learning about how the cloud works).

lsetiawan, 2020-10-28:
>I implemented a way to save to a cloud store simply by requiring a MutableMapping to a cloud store and passing that mapping into xarray/zarr. You can see how it would be used here: https://github.com/ngkavin/echopype/blob/class-redesign/echopype/tests/test_convert_ek60.py#L96

That implementation is great! I like the `to_zarr` approach for the syntax!

> It was done this way to remove dependencies on s3fs/gcfs/adlfs but echopype can still take in mappings created with these packages. Please let me know if something doesn't seem right (I just recently started learning about how the cloud works).

MutableMapping is great. Not quite related to this topic, what are your thoughts on input for the `Process` side of things? Should this also be some sort of MutableMap?

ngkavin, 2020-10-29:
@leewujung is currently redesigning how the `Process ` structure would look, but I think we will have a variable that points to the location of the backscatter/Sv/TS information that will be processed. That pointer should ideally be able to point to a file on the local machine, a place in memory, or to a cloud store.
Like `Convert` the pointer would only be a MutableMap if the file is located in the cloud, although it might be a good idea to have all pointers be maps with keys that indicate where the files are located, locally or otherwise.

emiliom, 2020-12-24:
FYI, I used @lsetiawan's PR #216 (well, the branch used in the PR) to re-run my notebook on pangeo that processes all the 2017 pacific hake survey GPS datagrams. Both as a test and as a performance comparison to the same run (sort of) with echopype 0.4.1 code. It's not a robust exercise b/c it compares one run with the current code with one run with the new code. And there's another difference: In the current echopype code (latest release), the NMEA messages/datagrams used are 'GGA', 'GLL' and 'RMC', whereas in Don's branch based off the class-redesign branch, it's been changed to only use 'GGA'. With those caveats:

- 2 hrs 4 min vs 3 hrs 15 min
- 5,132,314 points vs 7,700,028 points
- WAY less noise, bad locations

Assuming that difference in performance is representative, we can't tell how much of it is due to the direct S3 reads (rather than downloading to a file first, then reading the save raw file) vs the selection of fewer GPS datagrams (GGA only) and therefore fewer points. @leewujung I'd be curious to hear why only GGA messages are now used, but that's for a different issue, not this one.

leewujung, 2020-12-24:
1 hr difference in time! I wonder if it's saving from downloading files alone.

The NMEA message is definitely another point of discussion, which I mentioned in #210 . The short answer is that I found that the other messages contain low resolution data compared to GGA for the SH1707 EK60 dataset, and that these low-res messages are not in the other EK60 files I have, so I added that default but left the option to use others.

emiliom, 2020-12-24:
> The NMEA message is definitely another point of discussion, which I mentioned in #210 . The short answer is that I found that the other messages contain low resolution data compared to GGA for the [...]

Thanks for that follow up and reminder about #210. Let's follow up on this elsewhere.

emiliom, 2020-12-24:
@lsetiawan in #216 you said the PR was a "partial" solution to this issue. As far as I can see, it's a complete solution, and we can close this issue. Do you disagree?

lsetiawan, 2020-12-24:
@emiliom I was thinking this issue was to address the whole pipeline not just being able to access file-like objects. So yea, I think it's safe to say that we should close it. However, this is currently in a separate branch than master, so maybe we should wait?

emiliom, 2020-12-24:
> I was thinking this issue was to address the whole pipeline not just being able to access file-like objects.

FYI, there was an issue (#150) already open for the other end of the pipeline (writing to Zarr on object storage); just now I copied @ngkavin's update from two months ago to that issue, in preparation to be able to close this one.

> So yea, I think it's safe to say that we should close it. 
:fireworks:  

> However, this is currently in a separate branch than master, so maybe we should wait?

Yeah, I thought about that ... Let's keep it open, then. And thanks again!! This new capability is fantastic!

leewujung, 2020-12-24:
Chiming in on the closing issue bit: I have no bandwidth for echopype today, but in the next few days I'll be working to thoroughly check class-redesign on the conversion and processing side and merge that branch to master. Since @emiliom has checked #216 I think we can safely close this issue then.

emiliom, 2020-12-24:
Heh, I'm officially taking the day off today :smile: . It's Don's fault I'm working on echopype, b/c I didn't want to delay merging his awesome PR after he added the last commit. Then one thing led to another ... Anyway, I'll close the issue. I love closing issues!

closed 2020-12-24 by emiliom