echostack-org/echopype #202: What are the units for range_bin?

cramjaco, 2020-10-28:
I am working with EK60 data from the UNOLS vessel Siquliaq. After converting the .raw files to netcdf, I see that all of the dimensions have reasonable units, except for range_bin, which has no units as far as I can tell. These don't seem to be in meters, nor tenth of meters. Is there any advice for estimating how range_bin relates to depth? Thanks for any suggestions.

cramjaco, 2020-10-28:
On closer reading I see there are a bunch of processing steps. Presumably, one of these will give me a depth... I'll need to work with them a bit more.

cramjaco, 2020-10-28:
Processing doesn't seem to give me depths either... Any guidance for how to go from range_bin to depth would very much help. Thanks.

leewujung, 2020-10-28:
`range_bin` is unitless because it is simply the number of sample bins. You can covert it to the actual range in meters using other parameters in the .nc file. The [`calc_range`](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/process/ek60.py#L100-L107) method should have the elements you need if you read in the `Beam` group in xarray.

cramjaco, 2020-10-28:
This seems to work. For the record, I loaded a file in with `ed = Process(filepath)`, got the range data with `loc_cr = ed.calc_range()` and then aved out with `loc_cr.to_netcdf("rangedata.nc")`.

This gave me a netcdf file with the range data in it. I can then use that in combenation with the original netcdf file in my downstream analyses to assign depths to each range bin.

This would be easier if I could somehow add the depth data to the original netcdf file, is there a simple way to do that that I am missing? If not, my current approach will work for me.

leewujung, 2020-10-29:
I think you can just make the calculated range a dimension in the xarray dataset, and do swap_dims to align the observation with `range` instead of `rang_bin`, and export to netCDF afterwards if that's your next step.

closed 2020-11-01 by cramjaco