echostack-org/echopype #203: Do the processing steps in the documentation actually run in series, as written?

cramjaco, 2020-10-28:
I have a question about processing data files. Per the documentation one would process a netcdf as follows.

```
from echopype import Process
nc_path = './converted_files/convertedfile.nc'  # path to a converted nc file
ed = Process(nc_path)   # create a processing object
ed.calibrate()           # Sv
ed.remove_noise()        # denoise
ed.get_MVBS()            # calculate MVBS
```

It looks to me like each step does the processing but then doesn't save the data anywhere.  Thus `ed.get_MVBS` works on a file that hasn't passed through `ed.calibrate`, right? Or do these functions somehow mutate `ed` in place? I would expect that applying all three operations to ed would actually look something like this:

```
from echopype import Process
nc_path = './converted_files/convertedfile.nc'  # path to a converted nc file
ed = Process(nc_path)   # create a processing object
ed_cal = ed.calibrate()           # Sv
ed_den = ed_cal.remove_noise()        # denoise
ed_mvbs = ed_den.get_MVBS()            # calculate MVBS
```

And then I'd somehow save out ed_mvbs, perhaps by replacing the last line with `ed_den.get_MVBS(save = TRUE)` or maybe with something like `ed_mvbs_to.netcdf("output_processed.nc")`

Am I missing something? Does the code listed in the documentation actually provide the three steps, in sequence to the data in `convertedfile.nc`?

ngkavin, 2020-10-31:
`calibrate` can save a calibrated file into a file on your local machine, or hold the calibrated data in memory (which is the default).  Calling `get_MVBS` will get the Sv data if it is in memory, or you can specify a source path as a function argument. So in the example, everything is being saved in memory, and the functions are performed sequentially.

cramjaco, 2020-11-01:
Got it. Thank you.

closed 2020-11-01 by cramjaco