echostack-org/echopype #213: Error when running `to_netcdf` or `to_zarr`

lsetiawan, 2020-12-18:
## Overview

Within the `class-redesign` branch, when running `to_netcdf` I get the error below.

[image]

## Troubleshooting
1) Seems like there's no catch for when `self.convert_obj.ping_data_dict['angle']` has none channels ...

```
defaultdict(<class 'list'>, {1: array([[[  -1,   -1],
        [  -2,   -2],
        [  -2,   -2],
        ...,
        [ -78,   33],
        [ -66,   47],
        [  93,  -93]],

       [[  -1,   -1],
        [  -2,   -2],
        [  -2,   -2],
        ...,
        [  99,  -60],
        [-122, -125],
        [-125,  120]],

       [[  -1,   -1],
        [  -2,   -2],
        [  -2,   -2],
        ...,
        [  82,  -71],
        [-119,  -62],
        [  99,  -69]],

       ...,

       [[  -1,   -1],
        [  -2,   -2],
        [  -2,   -2],
        ...,
        [ -66,  -58],
        [  68,    5],
        [  99,  -65]],

       [[  -1,   -1],
        [  -2,   -2],
        [  -2,   -2],
        ...,
        [  74,   87],
        [  53,   77],
        [ -54,   79]],

       [[  -1,   -1],
        [  -2,   -2],
        [  -2,   -2],
        ...,
        [  38, -123],
        [  85, -122],
        [-122,  -17]]], dtype=int8), 2: None, 3: None})
```

2) I tried using the master just to make sure it's not the raw data, and sure enough, it's problem with the code.

Please help! @ngkavin @leewujung

leewujung, 2020-12-18:
Ah! The check must have been removed at some point...  Thanks for reporting this bug!

This is related to whether the transducer is single beam and split beam. Single beam files would not have the angle data. 
The configuration has a field "beam_type" where 0 = Single, 1 = Split.

@ngkavin : could you add a check for this? 
Some info below (if you're in SetGroupsEK60/EK80):
```python
# EK60
for ch in self.convert_obj.config_datagram['transceivers'].keys():
    self.convert_obj.config_datagram['transceivers'][ch]['beam_type']

# EK80
for ch in self.convert_obj.config_datagram['configuration'].keys():
    self.convert_obj.config_datagram['configuration'][ch]['beam_type']
```

lsetiawan, 2020-12-18:
Thanks @leewujung I'm really close to being able to read raw files from anywhere :smile:

leewujung, 2020-12-18:
Fantastic! You could just throw in that fix based on the above if you want. It's very small. :P  (I am just trapped in meetings at the moment)

lsetiawan, 2020-12-18:
I'll just let @ngkavin make that fix.. Looks like my stuff doesn't get affected, other than failing when trying to write. Reading is good so far. :+1:

emiliom, 2020-12-18:
> Thanks @leewujung I'm really close to being able to read raw files from anywhere

Awesome! I can test it next week if you'd like, if it's ready for testing.

lsetiawan, 2020-12-19:
@emiliom Have at it at #216 :smile:

leewujung, 2020-12-19:
Reopen since this is not merged to master yet.

commit f3e4ca4b (2020-12-19): Merge pull request #215 from ngkavin/class-redesign

This closes #213 .
closed 2021-04-27 by leewujung