echostack-org/echopype #222: Deprecation warning for API change

leewujung, 2020-12-22:
In 0.5.0 there will be major change in how the Convert and Process API are called, and it'll be good to add deprecation warning in this transition. We should be able to do this pretty easily for Convert. For Process let's try our best, but there will likely be breaking changes.

ngkavin, 2021-01-07:
Addressed in by #230

lsetiawan, 2021-01-07:
Looks like OOI is using echopype for their stuff now. We need to make sure that whatever echopype functions that they're using are covered. Just a reference here: https://github.com/oceanobservatories/ooicgsn-data-tools/tree/master/zpls_echograms

leewujung, 2021-04-25:
This is partially addressed in #282 by deprecating `Convert`.
Deprecation of `Process` is also in #296.

leewujung, 2021-05-05:
#304 deprecates the old `Process`. 
#330 covers the code changes in documention.

closed 2021-05-05 by leewujung