echostack-org/echopype #223: Documentation for cloud-interfacing libraries

leewujung, 2020-12-22:
We use fsspec for handling interfacing with various cloud platforms and won't include libraries like s3fs or gcsfs into echopype, but we should add detailed usage to the doc.

emiliom, 2021-01-11:
Have there been commits in the `class-redesign` branch (or another branch) involving changes to the documentation? I want to know whether to use master as the starting point for making documentation updates.

Also, FYI, enhancing the documentation to add the new cloud-interfacing functionality will also entail a broader update that are in the class redesign, such as the changes in #148.

emiliom, 2021-02-26:
Implemented in #256, but still in review and revision. See also https://github.com/OSOceanAcoustics/echopype/issues/243#issuecomment-775690393

emiliom, 2021-03-01:
In the new documentation for file conversion (https://doc-test-echopype.readthedocs.io/en/latest/convert.html), I've presented three schemes for S3 I/O: anonymous read, passing accesskey + secretkey, and opening a session from a `~/.aws/credential` file. The test CI uses Minio, and currently it only includes the second scheme (accesskey + secretkey). I looked around briefly to see whether Minio supports the other two schemes and couldn't find anything.

Separately, I've tested anonymous access (and it was also done in a previous version of `test_convert.py`), and unsurprisingly, it works. But b/c I don't have access to raw files on S3 that require credentials, I've never tested the session/credential-file scheme. There's no reason to think it won't work, but I wanted to document that it's untested, manually or on the CI.

I'll go ahead and remove the interim note on the docs that I added next to the session-access information.

leewujung, 2021-05-15:
I think we are good with this now with #337 merged to master. I'll go ahead to close this but @emiliom feel free to reopen if that's not the case.

closed 2021-05-15 by leewujung