echostack-org/echopype #229: Support pathlib Path in Convert

emiliom, 2020-12-24:
I've grown very fond of the convenience and readability of `Path` paths (see links below). In the current release (0.4.1), `Convert` supports the use of [`pathlib`](https://docs.python.org/3/library/pathlib.html) `Path` as file paths, in addition to strings and list of strings. But in the `class-redesign` branch that will soon be merged into `master`, that support was dropped inadvertently. It was previously supported only because `Convert` only supported local file paths and it did so using the standard Python file handling mechanisms which already support `pathlib` `Path`.

The issue is with the instance checks here:
https://github.com/OSOceanAcoustics/echopype/blob/class-redesign/echopype/convert/convert.py#L153
This would be fixed by changing that line to `if isinstance(rawpth, (str, Path)):`, after adding this import statement: `from pathlib import Path`. However, I have not checked if there are other changes that would need to be made.

@lsetiawan had this comment:
> I have not used `Path` before. What is the advantage of this and what are the use cases? If we do allow Path types, then this means that we have to check if all the packages downstreams also take this. At the end of the day, if the usecase of people using `Path` is small, maybe we should just put in the docs to encourage people to use string.

For the advantages, see below. Briefly, with pathlib the code is almost always more compact and more readable than with the alternatives (the usual combination of os.path and friends).

I'm not worried about downstream support. Pretty much all major libraries support it transparently these days. ie, wherever a file path string is accepted, a file path `Path` is equally accepted.

- https://treyhunner.com/2019/01/no-really-pathlib-is-great/
- https://treyhunner.com/2018/12/why-you-should-be-using-pathlib/
- https://rednafi.github.io/digressions/python/2020/04/13/python-pathlib.html
- https://www.digitalocean.com/community/tutorials/how-to-use-the-pathlib-module-to-manipulate-filesystem-paths-in-python-3

lsetiawan, 2020-12-24:
Looks like fsspec doesn't support this atm (see https://github.com/intake/filesystem_spec/issues/434), so we'd have to think about how this should be implemented fully within echopype for all the various use cases. Anyone know if echopype works on Windows? This whole thing about pathlib got me thinking of file paths from a windows OS.

emiliom, 2020-12-24:
Just when I had stopped working for the day, you come in and add interesting comments :stuck_out_tongue_winking_eye:. Nice fsspec thread, @lsetiawan ! It includes a reference to this promising package, https://github.com/drivendataorg/cloudpathlib, its [initial proof-of-concept fsspec integration](https://github.com/drivendataorg/cloudpathlib/pull/109), and a [sustained exchanged about fsspec](https://github.com/drivendataorg/cloudpathlib/issues/96).

However, my understanding is that your integration of fsspec into echopype involves separating out local (POSIX, Windows paths) file interaction such that those are handled by "traditional" means and not by fsspec. `pathlib` doesn't handle url's, as far as I can tell. So, I'm only proposing `Path` instances be accepted for local paths.

leewujung, 2020-12-24:
> Anyone know if echopype works on Windows? 
@ngkavin uses Windows OS, so we know the answer. 😄

lsetiawan, 2020-12-24:
> Just when I had stopped working for the day, you come in and add interesting comments

There is no getting away from echopype haha Just like an echo it always comes back :stuck_out_tongue: 

> So, I'm only proposing Path instances be accepted only for local paths

Okay that makes more sense. Haha This whole time, I thought pathlib for everything! :smile:

leewujung, 2021-04-27:
Alright, we've gone through this path I think and also in #293 @ngkavin @lsetiawan identified and figured out the best we can do at the moment given `Path`'s limitation on handling remote paths. 

I think we can close this now, but I'll defer to @emiliom who raised the issue. :)

emiliom, 2021-04-29:
This is good to go. I'll just add a couple of parting comments.

- `test_convert.py` uses a good mix of `Path` and string local file paths, so it looks like there are enough tests there. Though not really any dedicated units tests, as far as I can tell. Not a big deal, IMHO.
- In `test_convert.py` (and I'll guess the same for actual echopype code), the use of `Path` doesn't take full advantage of all the automatic castings and simpler syntax Path affords, so it looks a lot like it's reusing `os.path` patterns. Reminds me of C/C++/FORTRAN programmers coding Python for loops before they discover array operations and list comprehension :wink:. Also not a big deal, but we/me should `Path`-ify `test_convert.py` down the line
- "`Path`'s limitation on handling remote paths": I suspect remote paths (uri's) are out of scope for Path. In the future, to bring Path-like convenience to uri's, let's keep track of the promising package [cloudpathlib](https://github.com/drivendataorg/cloudpathlib), its [initial proof-of-concept fsspec integration](https://github.com/drivendataorg/cloudpathlib/pull/109), and a [sustained exchanged about fsspec](https://github.com/drivendataorg/cloudpathlib/issues/96).

closed 2021-04-29 by emiliom