echostack-org/echopype #232: Add NMEA message type as another data variable in Platform group

leewujung, 2021-01-01:
Currently EK data are default to only save the GGA NMEA messages in the Platform group. It will be better to save _all_ messages and include the message type as another data variable. See #231 for discussion.

emiliom, 2021-01-03:
I don't know anything about the different NMEA messages, what each of them means, or why they have different resolutions. So, I don't know whether the best default is to save only GGA vs all messages (*). But adding a NMEA-GPS-message-type variable is a good idea independently of what the default it. Plus, I also like the current functionality that gives users the choice of what message types to save.

(*) Actually, I take it back! From the perspective of a lossless conversion from raw to netcdf/zarr format, it's best to have the default include all data.

leewujung, 2021-04-27:
This can now be closed now that `class-redesign` is merged to `master`. :D

closed 2021-04-27 by leewujung