echostack-org/echopype #234: Enable versioning of the readthedocs documentation

emiliom, 2021-01-11:
It doesn't look like the documentation is *versioned*. Meaning, documentation associated with specific echopype releases are not made readily available via a link such as https://echopype.readthedocs.io/en/v0.4.2/. Given the major changes in 0.5.0, I think it'd be helpful to current users to ensure the current documentation remains available via a version link. I bet that's a readthedocs configuration. @leewujung does that ring any bells? I have somewhat recent experience with readthedocs [managing a package that does have versioned docs](https://ulmo.readthedocs.io/en/v0.8.5/), so I can look into it.

emiliom, 2021-01-11:
@leewujung can you add me as a mainainer on readthedocs, https://readthedocs.org/projects/echopype/? My profile name there is `emiliom`.

emiliom, 2021-01-12:
I have readthedocs admin access now, thanks! Unfortunately the setting I was hoping would do the trick of versioning docs (Admin > Advanced Settings > Global settings > Single version) is not it. Oh well, I'll look closer.

These links are for my own follow up:
- sphinx `conf.py`
  - echopype: https://github.com/OSOceanAcoustics/echopype/blob/master/docs/source/conf.py
  - ulmo: https://github.com/ulmo-dev/ulmo/blob/master/docs/conf.py
- `.readthedocs.yml`
  - echopype: https://github.com/OSOceanAcoustics/echopype/blob/master/.readthedocs.yml
  - ulmo: https://github.com/ulmo-dev/ulmo/blob/master/.readthedocs.yml

emiliom, 2021-01-12:
Ok, I don't know quite how it happened, but it looks like versioning *is* enabled now:
https://echopype.readthedocs.io/en/v0.4.1/
I made just one settings change (set `Show version warning` to checked) that I didn't think would have any impact on this, but now v0.4.1 is available! We'll see what happens when v0.5.0 is released.

leewujung, 2021-01-12:
I think I know :)  Under "Versions" I activated v.0.4.1 (if you scroll down you'll see the list of release tags). The tags are automatically included, but I forgot how I set that up, or if this was automatic. I actually did it when we were still in the meeting, but it took time to trigger the build so I didn't know if that worked already. You can see the time under "Builds." By default a build is triggered by new pushes, but you can also manually trigger a build.

emiliom, 2021-01-12:
Ah. I saw that build timestamp and was puzzled. That explains it!

> By default a build is triggered by new pushes

Yes, that's what I expected, but it hasn't been triggering builds for versions (releases); the only builds that are automatically triggered are tagged as "latest" or "stable". Anyway, my immediate concern was ensuring the docs for version 0.4.1 were available going forward, and that's already met. We can generate manual builds for specific versions as needed, until we get around to automating the version/release builds.

leewujung, 2021-01-14:
> it hasn't been triggering builds for versions (releases)

Triggering based on releases is something that can be configured in the Webhook section here, we just need to check that. Currently it is set to trigger on pushes or tags on master, that's why RTD actually does have the previous releases - which have tags) that can to be activated.

I just activated under automation rules to "active version with tag", which should do what you're asking for. We'll see once we have a new tag. ;)

leewujung, 2021-05-15:
We'll see if this works out once we push the tag v0.5.0.

leewujung, 2021-05-17:
Confirming this is working now 😺

closed 2021-05-17 by leewujung