echostack-org/echopype #235: Raw files with repeated pings raises error in SetGroups

ngkavin, 2021-01-13:
If a ping is duplicated in the raw file, an error is thrown when trying convert. Specifically, when merging datasets along the frequency dimension like in  `set_env` of `SetGroupsEK60`.
The following error is raised: `ValueError: cannot reindex or align along dimension 'ping_time' because the index has duplicate values`

This issue is mentioned here https://github.com/pydata/xarray/issues/2108
An example EK60 raw file that produces this error can be found here: https://rawdata.oceanobservatories.org/files/CE02SHBP/MJ01C/07-ZPLSCB101/2019/12/01/CE02SHBP-MJ01C-07-ZPLSCB101_OOI-D20191201-T000000.raw

imranmaj, 2021-08-12:
Potential solutions:

- Throw out duplicates
- Increment duplicates by smallest possible time amounts (e.g., 1 ns, 1 ms)

imranmaj, 2021-09-30:
Fixed in https://github.com/OSOceanAcoustics/echopype/pull/433

PR #433 (2021-09-23): Correct duplicate ping_times during EK60 conversion (fixes #235)
Fixes #235 

Corrects duplicate `ping_time`s during EK60 conversion by incrementing duplicate times by 1ns
commit 33c96c3f (2021-09-23): Correct duplicate ping_times during EK60 conversion (fixes #235) (#433)

* Add fix for duplicate ping_time

* Remove dead code

* Increment by 1ns instead of 1ms

* Add duplicate ping time warning and store in provenance

* Autoformatter

* Add test for duplicate ping times file

* Add qc method to remove duplicate pings

* Drop entire pings with duplicate values

* Store entire original ping_time and add attribute when there are duplicate ping_times

* Remove duplicate ping_time qc method, it will be added back in future PR

* Change EchoData combine ping_time reversal attribute to netCDF enc [... cut]
commit 62c87e9f (2021-10-01): Release/v0.5.4 (#448)

* re-attach sonar_model to combined echodata object (#437)

* Correct duplicate ping_times during EK60 conversion (fixes #235) (#433)

* Add fix for duplicate ping_time

* Remove dead code

* Increment by 1ns instead of 1ms

* Add duplicate ping time warning and store in provenance

* Autoformatter

* Add test for duplicate ping times file

* Add qc method to remove duplicate pings

* Drop entire pings with duplicate values

* Store entire original ping_time and add attribute when there are duplicate ping_times

* Remove duplicate ping_time qc method, it will be added ba [... cut]
closed 2021-09-30 by imranmaj