echostack-org/echopype #251: Default locations of converted file differ for local and model sources

leewujung, 2021-01-31:
Currently the default locations of converted files differ for local and model sources.
- For local sources: the converted files by default are saved into the same folder as the source files.
- For remote sources: the converted files by default are saved to `./temp_echopype_output`.

This is confusing and we should just pick one. 
@lsetiawan @emiliom @valentina-s : I think we decided to go with the route "converted files by default are saved to `./temp_echopype_output`" in the last meeting.

This is possibly leftover from #240 .

emiliom, 2021-02-01:
> I think we decided to go with the route "converted files by default are saved to ./temp_echopype_output" in the last meeting.

Gosh, I think so, but I forget. Regardless, that makes sense to me, with the possible exception that it'll be a change in behavior for existing echopype users who have not been specifying a save path.

lsetiawan, 2021-02-01:
Hmm.. okay. Right now I do have warnings for not specifying save path and also show where things are saved at. See below.

https://github.com/OSOceanAcoustics/echopype/blob/5eb200509ceb038a3649c9a8e19fdf045893a16c/echopype/convert/convert.py#L244

https://github.com/OSOceanAcoustics/echopype/blob/5eb200509ceb038a3649c9a8e19fdf045893a16c/echopype/convert/convert.py#L263

lsetiawan, 2021-02-01:
> Regardless, that makes sense to me, with the possible exception that it'll be a change in behavior for existing echopype users who have not been specifying a save path.

I figured, by having the warnings that shows where things are saved, it keeps the behaviour similar for existing users.

leewujung, 2021-02-05:
> I figured, by having the warnings that shows where things are saved, it keeps the behaviour similar for existing users.

This is fine, but my point is that we need to decide whether to save the outputs to `./temp_echopype_output`. Currently for local files it is to where the source file is.

lsetiawan, 2021-02-05:
> This is fine, but my point is that we need to decide whether to save the outputs to ./temp_echopype_output. Currently for local files it is to where the source file is.

Sure why not to standardize things, as long as we print out where it's saving, existing user should be able to know.

leewujung, 2021-02-05:
Yeah I think we should put the outputs in `./temp_echopype_output` if users don't specify `save_path`.

@lsetiawan : could you please make this change? 

@emiliom : could you please add a `.. warning::` block for this in your doc updates (along with all the other things from `raw2nc`/`raw2zarr` to `to_netcdf`/`to_zarr`)? 

Thanks both!

lsetiawan, 2021-02-05:
> could you please make this change?

Surely :smile:

emiliom, 2021-02-05:
> could you please add a .. warning:: block for this in your doc updates (along with all the other things from raw2nc/raw2zarr to to_netcdf/to_zarr)?

Will do

leewujung, 2021-04-27:
I think we are good on this. Closing this with #298.

closed 2021-04-27 by leewujung