echostack-org/echopype #255: Nuntype and reshape error in raw2nc() for EK80 data

EspenWinther, 2021-02-05:
I am trying to open some of my own datasets and was wondering if there are some limitations I am not aware of. I am getting errors "Message='NoneType' object is not subscriptable" for some datasets and "ValueError: cannot reshape array of size 30934 into shape (30934,newaxis,4)" for another. Both datasets are captured the same day with the same equipment using the EK80 and a splittbeam echosounder. Does anyone have any idea why this may be? the files are rather large so perhaps it is a size issue?

Just testing simple code:
dc = Convert(path='convert_test.raw', model='EK80')  # for EK80 file
print(dc)
dc.raw2nc()

leewujung, 2021-02-11:
@EspenWinther : how many channels do you have, and how big are the files? The one thing I know we have not looked into to ensure compatibility is Simrad's newest addition of the mission planner. Though your error looks to be more like the "shape" of the data. If the file size is a problem it most likely won't error out at that stage.

Could you send us an example file? A link here or in an email would work.

EspenWinther, 2021-02-12:
Thanks for your quick reply! I need to investigate if I am at liberty to send any of the files that I have, however it strikes me that perhaps it might be due to the echo-sounder used perhaps? I am using an ES70-7CD splitt beam echosounder with four sections. Could this be the issue? the file I am using as a test case here is 102 704 KB, so perhaps not large enough to cause any issues

leewujung, 2021-05-19:
@EspenWinther : Could you try to see if this problem got resolved by using echopype v0.5.0? Note that we revamped the API and some underlying code structure, but without an example data file it is hard to say if the issue you ran into is covered.

leewujung, 2021-11-27:
Closing this since it's stalled for a long time and we don't have enough info or test files to know the cause of such a problem.

closed 2021-11-27 by leewujung