echostack-org/echopype #263: Prune test data files on Google Drive folder used in the GH CI

emiliom, 2021-02-25:
PR #260 introduces downloading (via `rclone`) of test data from a Google Drive folder, for use in GitHub Actions CI tests. The core of the code that downloads the files will be at `.github/actions/gdrive-rclone/entrypoint.sh`. Currently the only tests that are enabled are in `echopype/tests/test_convert.py`. The Google Drive folder contains both the data files used directly in those tests, and older data files not used there. Those other files are largely what's currently in `echopype/test_data`, including the `from_matlab`, `from_echoview` and `ncei-wcsd` sub-folders, and other miscellaneous files. 

To speed up the GitHub Actions CI, the `gdrive-rclone` action should download *only* what's actually used in tests. I suggest that the extraneous test data files not currently being used as part of CI tests should be removed from the Google Drive folder downloaded by `gdrive-rclone`.

leewujung, 2021-02-26:
The folders `from_matlab`, `from_echoview` are for testing the actual parsed values against those from other software, so we cannot remove them. There are many other tests in addition to reading from different source locations. 

I don't know what `ncei-wcsd` is and we can remove it unless someone else knows what that is for.

emiliom, 2021-02-26:
I'm only referring to the Google Drive folder (and set of files) that's currently rclone'd in order to run the CI tests currently in the GH action. I'm not suggesting the other files be removed from the GitHub repo (`echopype/test_data` folder) or anywhere else on Google Drive. Currently the only test in the GH CI action Don has set up is `echopype/tests/test_convert.py`, and I don't know what other tests currently in the `echopype/tests` folder are intended to be added to the GH CI in the very near term. As other tests are added to the CI, the test files they need can be added to the Google Drive folder; until then, having those files there is just adding to the CI processing time.

lsetiawan, 2021-02-26:
The test folder that I'm currently using for the `test_convert.py` is in `ncei-wcsd` folder. We can reactivate tests as needed. Maybe we can have a separate google drive folder that we manage to store the non-active files?

leewujung, 2021-10-30:
Seems that this has been resolved during all this time? :) The current CI runs through all the tests using files in the Google Drive.

closed 2021-10-30 by emiliom