echostack-org/echopype #264: remove file download code from test_convert.py?

emiliom, 2021-02-26:
In [echopype/tests/test_convert.py](https://github.com/OSOceanAcoustics/echopype/blob/class-redesign/echopype/tests/test_convert.py), [_download_file()](https://github.com/OSOceanAcoustics/echopype/blob/class-redesign/echopype/tests/test_convert.py#L55) and [download_files()](https://github.com/OSOceanAcoustics/echopype/blob/class-redesign/echopype/tests/test_convert.py#L95) are no longer used now that the test data files are all staged via Google Drive + rclone + Docker. Should this "download" code be removed?

leewujung, 2021-04-28:
@emiliom : just wanted to ping to bring this back to the foreground again now that we have a cleaned-up `master` branch and all tests are running. Thanks!

emiliom, 2021-04-29:
As far as I can tell from inspecting both the [test_convert.py code](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/tests/test_convert.py) and the log it generates, [download_files()](https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/tests/test_convert.py#L95) (and hence `_download_file()) is not used. And I don't really see a scenario where we would use it, or something we actually want to test for involving that code. So, I vote to remove it.

@lsetiawan what do you think? If you agree, I can submit a PR.

emiliom, 2021-05-03:
Hey @lsetiawan, could you confirm what I said above :point_up_2: about `download_files()` and `_download_file()`?

PR #322 (2021-05-03): Remove download files code in convert test
This PR removes code that download files as convert.py is getting setup since now everything happens during docker build, and test files are actually copied from the http docker container.

Closes #264
closed 2021-05-03 by lsetiawan