echostack-org/echopype #266: Overhaul calibration code and API call pattern

leewujung, 2021-03-02:
This issue is long overdue (both in terms of making its way here and the action).

The major change from v0.4.1 to v0.5.0 (WIP) is to overhaul the underlying module structure for calibration (which is instrument-specific) so that it is easier to maintain and read now that we're expanding to more sonar models. We are also changing the API call pattern (see #250) to be more modularized wrt the functionality, rather than dump everything into the previous `Process` object.

The general module reorganization is in the `class-redesign` branch, whereas the specific overhaul of calibration is in the `echodata-overhaul` branch.

(I kept on needing a place to interim notes and finally figured out that's what @emiliom has been doing with the docs branch! 🤓)

emiliom, 2021-03-04:
> I kept on needing a place to put interim notes and finally figured out that's what @emiliom has been doing with the docs branch! 

It's worth noting that the docs lend themselves nicely to being in a separate branch, b/c the files are completely separate from everything else. Well, except for the API docs build, but I'm just ignoring all the error messages for now!

leewujung, 2021-03-08:
> except for the API docs build

🤯     
Yes, we need to do a housekeeping soon, once the calling patterns are settled and the tests are in place.

leewujung, 2021-04-27:
This can now be closed with `class-redesign` merged to `master`. Hurray!

closed 2021-04-27 by leewujung