echostack-org/echopype #268: LFS Issue

DLWangSan, 2021-03-18:
can not download your test files with LFS. Following is the log file, could you take a look please?
```
$ git-lfs.exe filter-process
Error downloading object: echopype/test_data/azfp/17031001.01A (32173cb): Smudge error: Error downloading echopype/test_data/azfp/17031001.01A (32173cb43dd6f10f9e33d3362edd36909409ffa9a3c4ad5111514ffc2906c381): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.
batch response
...
```

leewujung, 2021-03-18:
@DLWangSan : We are moving away from using LFS for storing the test data due to the cost associated with the bandwidth usage from reading EK80 test files. We are setting up another location for the test files in the upcoming release and will keep you posted when the release is out.

emiliom, 2021-04-06:
@leewujung can we close this issue and keep track of the Git LFS deprecation in #271?

leewujung, 2021-04-06:
Yes that's a good idea. I'll close this here.

closed 2021-04-06 by leewujung