echostack-org/echopype #269: Adaptation of new equipment (ES80) data

DLWangSan, 2021-03-18:
Hello, I used echopype to analyze the EK80 data on board, and the workload is greatly reduced! Now that we have ES80 sonar data, I wonder if subsequent versions of echopype will add support for ES80 devices?

The following is some data of ES80 sonar equipment：
```
https://drive.google.com/drive/folders/1j7GKaE-WpfAhrD3MHQ4PXe1yHo3PaFIb
```

Thank you !

leewujung, 2021-03-18:
@DLWangSan : glad that you found echopype helping your work! :)

It'll be awesome to include ES80 support in echopype, though at the moment we are short on hands to add support for ES80.

Would you be interested in helping adding the support for ES80?  A quick googling led to [this doc](https://www.simrad.online/es80/ref/es80_ref_en_a4.pdf) that explains the format and it seems reasonable. I can help with getting things into the standardized format.

DLWangSan, 2021-03-19:
thank you very much! If I can help, I would be happy to contribute a little to echopype

leewujung, 2021-03-22:
Great! If you are able to get started on a parser to read in the data, I can set up a branch that you could PR to. Just let me know.

gavinmacaulay, 2021-07-28:
As I understand it, the file format of the ES80 is identical to the EK80 (with a few trivial metadata changes).

leewujung, 2021-08-07:
@gavinmacaulay : that's great! Do you have some test files that we could try on?

gavinmacaulay, 2021-08-07:
I have ES80 files that I can offer. Where should I put them? I also have ES70 files (same format as the ES/ER60 files) if desired. 

To go with these, I have a nearly ready PR that makes the fairly trivial changes necessary for echopype to read ES70 files (ES80 will be just as trivial).

leewujung, 2021-08-07:
Fantastic! I just sent you an invitation to a Google Drive folder we use to store test data files at the moment (as documented [here](https://echopype.readthedocs.io/en/stable/contributing.html#test-data-files)). 

Looking forward to the PR!!  😃  We can add tests to run through your files once they are in the folder, like what we have for all the other formats.

gavinmacaulay, 2021-08-09:
echopype reads ES80 files without any changes needed if those files only contain data from WBT transceivers (the files are near identical to those produced by the EK80 software). But ES80 (and EK80) files can also contain data from a GPT transceiver. Currently, echopype fails to read those files. See #409 for some more details.

leewujung, 2021-11-27:
Closing this so that ES80-related issues are consolidated, in #497.

closed 2021-11-27 by leewujung