echostack-org/echopype #271: Are we fully deprecating use of Git LFS?

emiliom, 2021-03-28:
The documentation retains the [text about the use of Git LFS](https://doc-test-echopype.readthedocs.io/en/latest/contributing.html#test-files). But I think we've effectively stopped using LFS, right? If so, should I remove that text from the docs altogether?

ref: #268

leewujung, 2021-03-29:
Yes, I agree we should remove LFS text from the docs!

leewujung, 2021-04-25:
This issue can be closed one `class-redesign` is merged to `master`.

leewujung, 2021-04-27:
Closing this with #298.

leewujung, 2021-05-05:
We will need to remove all test files from the repo, and potentially "rewrite" the history to fully deprecate use of Git LFS (hopefully not needed!) to actually close this. But this does not impact releasing v0.5.0.

lsetiawan, 2021-05-05:
This may help us: https://github.com/git-lfs/git-lfs/issues/3026

leewujung, 2021-05-05:
Great, thanks for pointing to this @lsetiawan . Let's make sure to experiment with a duplicated dummy first 😅.

lsetiawan, 2021-06-01:
With #369 merged. When I do `git lfs ls-files` nothing shows up, I assume that means it's not stored in lfs.

leewujung, 2021-06-01:
When I do `git lfs ls-files` I still get a whole list of files, but we can probably live with that for now, as long as folks don't have to do pull files from git lfs when they try to contribute.

leewujung, 2021-06-01:
An additional note: @lsetiawan since you're on this, could you remove all the LFS related stuff under `.gitattributes`? Thanks.

lsetiawan, 2021-06-01:
> When I do git lfs ls-files I still get a whole list of files,

Are you on the branch associated with #369?

lsetiawan, 2021-06-01:
>could you remove all the LFS related stuff under .gitattributes? Thanks.

That should be all deleted now on dev branch.

leewujung, 2021-06-01:
Great, now under dev when i do `git lfs ls-files` I don't see anything.

leewujung, 2021-12-04:
Reopening this issue because there are still some hidden problems related to Git LFS, which show up when maintainers try to push to the PR forks (@lsetiawan and I both ran into this). The workaround has been to skip pre-hook as discussed [here](https://github.com/git-lfs/git-lfs/issues/2291) and we should probably fix the source of the problem. From other people's reporting it seems that this could cause problems to contributors also depending on what commits they pull.

Completely removing LFS from the repo appears to be a common need and a casual search turned up [this](https://bryanbrattlof.com/removing-git-lfs-from-a-project/). Not sure if there is a better solution. I did the "rewrite history" part when adding LFS to the repo. It is pretty terrible because people need to delete and refork, otherwise the history don't match.

@lsetiawan : thoughts?

leewujung, 2022-10-13:
Let's investigate this as part of the test overhaul in 0.6.4, but not delaying this further so that we can once and for all be happy that we don't have LFS constraints anymore.

leewujung, 2023-03-11:
I'll close this now but if problems surface later we'll reopen it.

closed 2023-03-11 by leewujung