echostack-org/echopype #286: Revamp tests to work with `open_raw` and misc fixes

leewujung, 2021-04-22:
With #282 now we have a stable and consistent pattern to access/use both raw and converted data, which is great, but that also leaves the original convert tests to revamp due to the API change. 

There are also some more granular changes @lsetiawan and I thought would be good to make changes for with the new `open_raw` code. 

Below is a checklist for tying up these loose ends. 

- [ ] change all instances with mixed use of `Path` and `os` to all use `Path` in `echopype/convert/api.py`, such as in [here](https://github.com/OSOceanAcoustics/echopype/blob/class-redesign/echopype/convert/api.py#L99) and [here](https://github.com/OSOceanAcoustics/echopype/blob/class-redesign/echopype/convert/api.py#L99)

- [ ] use `Path` in all cases where `os` is currently used on `echopype/convert/api.py`

- [ ] make [`to_file`](https://github.com/OSOceanAcoustics/echopype/blob/class-redesign/echopype/convert/api.py#L142) and [`_validate_path`](https://github.com/OSOceanAcoustics/echopype/blob/class-redesign/echopype/convert/api.py#L50) to work with just one file only. We can do the multiple files part when we get to make `open_mfraw`. For discussions on this see https://github.com/leewujung/echopype/pull/3

- [ ] change input arguments `file` --> `raw_file` and `model` --> `sonar_model` under [`open_raw`](https://github.com/OSOceanAcoustics/echopype/blob/class-redesign/echopype/convert/api.py#L428-L434) -- this has implications with all tests

- [ ] overhaul all tests in `test_convert.py` involving using the deprecated `Convert` object to use `open_raw` and the resulting `EchoData`

- [ ] overhaul all tests in `test_convert_....py` that currently save an output file before loading the data back in to compare with test datasets, to just compare the attributes of the returned `EchoData` object and the test datasets

@lsetiawan feel free to chime in on things I may have missed or modify the above.

Thanks @ngkavin !

lsetiawan, 2021-04-22:
@ngkavin There is a helper python script that will setup the services for testing. Run the command below on the root of echopype. Make sure you have `docker` and `docker-compose` installed. Let me know if you have any questions! Make sure to perform tests against these services.

### Deploying 
```
python .ci_helpers/docker/setup-services.py --deploy
```

### Tearing down
```
python .ci_helpers/docker/setup-services.py --tear-down
```

ngkavin, 2021-04-27:
Convert tests are now working largely thanks to @lsetiawan! All tasks have been completed, but the `os` module was re-added to the API due to `pathlib` not being able to work with URLs, incompatibilities with Windows Paths, and being unsupported by `fsspec`.

closed 2021-04-27 by ngkavin