echostack-org/echopype #307: Fix `frequency` encoding in AZFP conversion

leewujung, 2021-04-29:
It appears that `frequency` has been encoded in `int32` for AZFP converted files, which results into errors when computing for environmental variables #304 . Let's fix this by encoding it as `float` as specified in the convention.

leewujung, 2021-04-29:
Closed by #309 .

emiliom, 2021-04-29:
The same change will have to be made for the other sensor types (EK60, etc), right? How about expanding the scope of this issue (and reopening it) beyond AZFP, so the others are not forgotten?

leewujung, 2021-04-29:
Yes I was gonna add a comment in the convention issue. We can reopen this issue and do everything here. :)

leewujung, 2022-03-17:
I think we can probably make this in a minor release v0.6.X once we have the main (breaking) things down.

emiliom, 2023-05-23:
The change to float was done for AZFP in PR #309 (originally to the old `frequency` variable).

For EK60 and EK80, `frequency_nominal` is already set to float, too. 

I think we can close this issue.

leewujung, 2023-05-23:
Sound good. I imagine this will be covered by #1043

emiliom, 2023-05-23:
Thanks. Yeah, the goal in #1043 is to also check for data types.

I'll close this issue now.

closed 2023-05-23 by emiliom