echostack-org/echopype #325: Implementing convention checks for echopype dataset.

lsetiawan, 2021-05-03:
Following up on my pydantic ideas. I have started echopydantic, where I think we can really define the convention as python data models. https://github.com/lsetiawan/echopydantic

You can see it in action on this development notebook: https://github.com/lsetiawan/echopydantic/blob/main/notebooks/Development.ipynb

Still have to think about the implementation of this, regarding of how this mixes with xarray objects, etc.

_Originally posted by @lsetiawan in https://github.com/OSOceanAcoustics/echopype/issues/210#issuecomment-831428193_

lsetiawan, 2021-05-03:
My first attempt to handling these conventions is creating a yaml file that somewhat describe the convention in https://raw.githubusercontent.com/OSOceanAcoustics/echopype/e445cda7cec8ea703ec6bcd6205186ee9a463c82/echopype/echodata/convention/1.0.yml. I am currently using that file to dynamically create the attributes in EchoData. 

My thoughts on this is possibly using [pydantic](https://pydantic-docs.helpmanual.io/) in the future to describe the standardized EchoData attributes that somewhat adhere to the spec. This would actually help with the validating with minimal effort. But I'm still trying to wrap my head around how to integrate this.

_Originally posted by @lsetiawan in https://github.com/OSOceanAcoustics/echopype/issues/210#issuecomment-823426558_

leewujung, 2023-07-25:
Related to #1043