echostack-org/echopype #356: AD2CP ping_time values become inaccurate after saving and loading

imranmaj, 2021-05-17:
Saving and loading ping_time values leads to incorrect values reminiscent of floating point imprecision (e.g., saving a ping_time `2021-03-17T19:47:19.188500000` to disk and loading might produce `2021-03-17T19:47:19.188499999`).

Currently, AD2CP tests avoid this by saving data in memory instead of writing to disk:
[https://github.com/OSOceanAcoustics/echopype/blob/66fca2ba519ec00f80647582eff3684bdccf3c9a/echopype/tests/test_convert_ad2cp.py#L36-L38](https://github.com/OSOceanAcoustics/echopype/blob/66fca2ba519ec00f80647582eff3684bdccf3c9a/echopype/tests/test_convert_ad2cp.py#L36-L38)

imranmaj, 2021-05-17:
This is broken with xarray version v0.16.2. Changing xarray version to v0.17.0 fixes this (unknown if any versions earlier than v0.17.0 but after v0.16.2 will fix this).

lsetiawan, 2021-05-17:
I have ran into this issue before. This is why I created this method: https://github.com/OSOceanAcoustics/echopype/blob/66fca2ba519ec00f80647582eff3684bdccf3c9a/echopype/convert/set_groups_base.py#L41-L56. Interesting that this is still happening. I'll have to look into it. Thanks for reporting this @imranmaj.

lsetiawan, 2021-05-18:
Problem here is that `set_encodings`:https://github.com/OSOceanAcoustics/echopype/blob/66fca2ba519ec00f80647582eff3684bdccf3c9a/echopype/convert/set_groups_base.py#L59-L79

are not being used anywhere within `set_groups_ad2cp.py`. https://github.com/OSOceanAcoustics/echopype/blob/master/echopype/convert/set_groups_ad2cp.py

Currently the way it is, xarray is determining the encoding for the datasets.

lsetiawan, 2021-05-18:
@imranmaj please make the necessary to fix this issue. The `set_encodings` functions need to be implemented for the adcp set groups. It is an explicit way to ensure that `open_raw` and `open_converted` behaves the same way and data are not modified.

lsetiawan, 2021-05-18:
@leewujung Once @imranmaj makes the necessary changes, I think this could be a small bug fix that can be released as 0.5.1?

leewujung, 2021-05-18:
Yeah I agree this is a small bug fix that can go into v0.5.1.

PR #361 (2021-05-25): Fix AD2CP Encoding Issues
* Fixes data encoding issues outlined in #356
* Adds length check to _encode_dataarrays (to fix data encoding issues)
* Renames `time_*` dimensions to `ping_time_*`

Closes #356
closed 2021-05-25 by leewujung