echostack-org/echopype #360: Conversion improvements for AD2CP files

leewujung, 2021-05-19:
Let's use this issue to note down the various things to improve for the AD2CP file conversion. The issue text will continue to be updated.

Note that a lot of these are related to what encoding convention we'd follow since there are _a lot of_ ad-hoc addition to the current convention and that needs to be reviewed also.

- Under the `Beam_complex` group `echosounder_raw_transmit_samples_r|i` has the same `ping_time` dimension with NaNs padded to the same dimension as `echosounder_raw_transmit_samples_r|i`, because the transmit samples only contains 1 ping. Since there is a separate `time_echosounder_raw_transmit` coordinate, this NaN padding shouldn't happen?

leewujung, 2021-05-19:
@imranmaj : tagging you on this. Let's make a plan to fix these once we pick up the more immediate leftovers from v0.5.0.

imranmaj, 2021-06-17:
Spikes occur because data was being parsed as unsigned integers in accordance with the specification, but they should actually be parsed as signed integers

leewujung, 2021-07-18:
These issues are addressed by #388.

PR #388 (2021-07-17): AD2CP Parsing Improvements
* Uses `np.float64` explicitly for AD2CP numerical data
* Indexes AD2CP data variables by their correct `ping_time_*` dimensions instead of the overall `ping_time` dimension
    * Updates AD2CP tests to reflect this dimension change
* Changes AD2CP tests to use `np.allclose()` instead of `np.isclose().all()`
* Removes the AD2CP `velocity_echosounder` data variable from the `Beam` group
* Removes the AD2CP `amplitude_echosounder` data variable from the `Beam` group and renames the variable previously named `echosounder` to `amplitude_echosounder`
* Moves AD2CP raw ech [... cut]
closed 2021-07-18 by leewujung