echostack-org/echopype #368: Need docs on using pre-commit for contributors

lsetiawan, 2021-05-28:
@emiliom could you create some docs for contributors to be able to use `pre-commit`? Thanks!

emiliom, 2021-06-01:
Sure. But can you add a bit of information here about what it encompasses, exactly? I'm not confident I'm in the best position to describe it, but I can take run with it if you give me a starting point.

lsetiawan, 2021-06-01:
So, `pre-commit` is actually in `requirements-dev.txt`, a contributor would install all those packages there. Once all of that are installed. They should run `pre-commit install` before all the docker services deployment and stuff like that. This way, everytime they make changes to the code and try to commit, pre-commit runs and adjusts as needed before they can actually commit changes. If they want to manually run `pre-commit` before actually doing git commit, then they can also do `pre-commit run --all-files`. Here's a link to the pre-commit docs: https://pre-commit.com/#usage. Thanks @emiliom  😄

emiliom, 2021-06-16:
I've moved this issue to the 0.5.2 milestone. It'll be part of a larger batch of "contributing" doc updates that I'll work towards releasing to live (readthedocs `stable`) incrementally rather than with code releases.

emiliom, 2021-08-14:
Addressed by #417

PR #417 (2021-08-14): Contributing page docs udpates
- pre-commit hooks section (closes #368)
- update info on docker-compose
- reorganize and expand section on contributing to the documentation, and RTD versions
closed 2021-08-14 by emiliom