echostack-org/echopype #371: Update README in test_data to match README in google drive or vice versa

lsetiawan, 2021-05-29:
It appears that the README in test_data folder within the repo is different than the README in google drive. This needs to be fixed so that git doesn't think we changed the file.

leewujung, 2021-05-29:
I replaced the outdated README in the google drive.
However, is there a need to have that README in the google drive? It is not version controlled, and since the content of test_data is replaced entirely when tests are run, can we just remove it?

lsetiawan, 2021-06-01:
> It is not version controlled, and since the content of test_data is replaced entirely when tests are run, can we just remove it?
No we can't remove it since it is in echopype repo. Yes, it is replaced, however if we remove it, unless we gitignore the README, git will detect it as deleted since it's not in google drive.

leewujung, 2021-06-01:
Ok, let's keep it this way then. Just need to remember this when we finally update the test files.

lsetiawan, 2021-06-02:
Update: Just tried this out by doing `python .ci_helpers/docker/setup-services.py --deploy`. It filled the `echopype/test_data` folder with actual test datasets including README and git did not detect any changes! Closing ...

closed 2021-06-02 by lsetiawan