echostack-org/echopype #373: Add explanation of `range_bin` to docs

leewujung, 2021-05-29:
In the raw converted data we have the coordinate `range_bin` which is just sample numbers (0, 1, 2, ...). This coordinate is then converted to `range` or `depth` in physical units (meters) when the data is calibrated. This can be confusing to the users and we should expand this in the docs.

This is related to #259 -- we should provide a couple simple helper function for this conversion.

emiliom, 2021-06-01:
Related to this documentation, also need to add variable attributes to the `range_bin` coordinate variable, which currently doesn't have any. I can do this while adding to the documentation.

For reference, for EK60 here's where the code to add attributes would go (in the `dev` branch): [echopype/convert/set_groups_ek60.py#L528](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/convert/set_groups_ek60.py#L528)

emiliom, 2021-12-02:
@leewujung do you have a suggestion for where the `range_bin` description should go in the documentation? I'm thinking, for now, it belongs in https://echopype.readthedocs.io/en/stable/data-format.html#modifications-to-sonar-netcdf4

That section also needs to be rewritten, borrowing heavily from text and figures in the paper.

leewujung, 2021-12-03:
Yes I think it'll work well to have the description in the part you proposed, and agree with updating the whole thing using materials from the paper.

One thing it'd be nice to have is probably some printouts from an `EchoData` object to show how things look like exactly.

closed 2021-12-09 by emiliom