echostack-org/echopype #376: AD2CP raw data samples are imprecise

imranmaj, 2021-06-04:
Raw AD2CP data samples differ from expected Ocean Contour output by approximately 4*10<sup>10</sup>. Is there a way to eliminate this imprecision? Could this be caused by encoding issues/floating point error?

Currently in the process of emailing Nortek.

See [https://github.com/OSOceanAcoustics/echopype/pull/361#discussion_r637068682](https://github.com/OSOceanAcoustics/echopype/pull/361#discussion_r637068682)

imranmaj, 2021-07-01:
For a small set of samples, the difference between Ocean Contour parsed values and Echopype parsed values is shown in this histogram:
[image]
The signed fraction format used by the AD2CP raw samples has a precision of 1/2<sup>31</sup> (i.e., all possible numbers that can be represented in this format occur in intervals of 1/2<sup>31</sup>). Since the differences in the values is equal to or less than 4.66*10<sup>-10</sup> ≈ 1/2<sup>31</sup>, the raw sample imprecision can be attributed to imprecision in the data format.

closed 2021-07-01 by imranmaj

leewujung, 2021-07-01:
This is great! @imranmaj thank you for investigating this!