echostack-org/echopype #381: Matching time or location of env_params used in calibration

leewujung, 2021-06-14:
Currently the `cal_params` can be specified by user when calling calibration functions, but the values are restricted to be static values. It is a common requirements when analyzing data that we use the actual CTD data for the temperature, salinity, and depth to calibrate the data. The CTD data can be aligned with time or location (or both) with the acoustic data. We should be able to allow `cal_params` to be passed in as an xarray Dataset that can be automatically aligned during calibration.

leewujung, 2021-06-14:
@imranmaj : I'll get some actual CTD data from the hake survey data for this.

leewujung, 2021-06-28:
In terms of functionalities, the env parameters could probably come in in a few forms:
1. In the case of a mooring where we may have CTD profiles at specific times, we would want to align the env parameters along the `ping_time` coordinate by a 1D interpolation.
2. In the case of a survey using mobile platforms like ships, we probably want to default to aligning the the env parameters with the acoustic data along with the platform lat/lon. The lat/lon are likely data variables themselves with a time coordinate. The alignment would be interpolation in 2D using scatter data (say each scatter point are a collection CTD data at a specific lat/lon).
3. In the case of an organized, gridded CTD data product with 3 coordinates in lat/lon/time. The alignment would be interpolation in 3D.

@imranmaj : what do you think about putting in support for case 1 first since it is the most straightforward, and then we can work out the path to support case 2 & 3?

imranmaj, 2022-02-10:
@leewujung case 3 hasn't been implemented yet, should we keep this issue open for that case?

leewujung, 2022-02-10:
@imranmaj : Good point! How about if we (you 😁) open up another issue to keep track of that?