echostack-org/echopype #387: Updates to Contributing docs page

emiliom, 2021-06-16:
- Change description of git workflow used to development-branch workflow. Don shared the link to https://www.atlassian.com/git/tutorials/comparing-workflows/gitflow-workflow to reference the gitflow workflow, specially in reference to a release branch 
- docker-compose no longer needs to be pre-installed locally, #375
- Need docs on using pre-commit for contributors, #368
- New option to be able to run all tests in the CI on a PR, #357

leewujung, 2021-06-17:
We should also add somewhere in this page that the method docstrings follow the numpy doc style: https://numpydoc.readthedocs.io/en/latest/format.html

emiliom, 2021-06-17:
From @lsetiawan, regarding recommended git practices:

> Squash will make all commits to be one commit, while the regular merge will create another commit for merge as well as the commits that followed that. Merging to `dev` branch from PR's should be squashed so that each squashed commit highlights the specific feature that was added with a reference to the PR. While merging to `master` SHOULD NOT be squashed as we should include the feature commits that leads to that specific version of the released software. Does that makes sense? I know it's a lot to remember but it would keep our git history clean and makes more sense

emiliom, 2021-07-18:
Linking in a couple of related issues: #378, #397

lsetiawan, 2021-08-24:
To run all tests on PR, simply add label `Needs Complete Testing`. https://github.com/OSOceanAcoustics/echopype/labels/Needs%20Complete%20Testing

I don't remember if someone outside of the organization can add label or not. Might have to check on this.

emiliom, 2021-08-24:
Thanks @lsetiawan ! Can you now describe this?

> Describe the conditional testing CI scheme that's in place

lsetiawan, 2021-09-27:
> Describe the conditional testing CI scheme that's in place

The CI runs on each PR based on the changes that have been made within various modules in echopype as listed here using the `run-test.py` script: https://github.com/OSOceanAcoustics/echopype/blob/d9405a4ba7c4c8863f2ebe54abb8d8663e5d2cf0/.ci_helpers/run-test.py#L26-L34

It will only run the test for that particular module that have changed. To retrieve what has changed, we use the get-changed-files github action (https://github.com/lsetiawan/get-changed-files).

emiliom, 2021-10-01:
> The CI runs on each PR based on the changes that have been made within various modules in echopype as listed here using the run-test.py script: 

@lsetiawan I was going through `run_test.py` to understand this functionality. It looks like the ability to test only a subset of modules, where those modules are passed as arguments at run time, does not apply to local runs (when `--local` is used). The `touchedfiles` arg is only [handled](https://github.com/OSOceanAcoustics/echopype/blob/main/.ci_helpers/run-test.py#L72) when `args.local` is False. Assuming I'm getting this right, it looks like it could be modified fairly easily for a local context by checking if `args.touchedfiles` is not empty. That would be handy! I might try it out later -- right now I'm only working on documentation.

emiliom, 2021-10-02:
@lsetiawan and @leewujung :

I've updated the Contributing page; [take a look](https://echopype.readthedocs.io/en/stable/contributing.html), and if you find something to fix that's relatively straightforward (won't take you a lot of time) I suggest you just use the "Edit on GitHub" link to create a PR. Just remember to have the PR merge into `stable`.

As far as additional updates to the Contributing page at this time, I'd like to add a bit about this:

> Squash will make all commits to be one commit, while the regular merge will create another commit for merge as well as the commits that followed that. Merging to `dev` branch from PR's should be [...]

To make sure I follow, what this refers to is what's described [here](https://www.git-tower.com/learn/git/faq/git-squash/), right? And, how are you doing this in practice? It sounds like the easiest and cleanest is to merge the `dev` into `main` via a PR in GitHub, while selecting the "Squash and Merge" option :
[image]

@leewujung is that what you've been doing?

emiliom, 2021-12-02:
The last task was addressed in #482. Ready to close this issue

PR #482 (2021-12-02): docs: git and PR merging workflows, output_storage_options docs [skip ci]
- Convert raw files: Describe use of `output_storage_options`
- Contributing to echopype: Elaborate on the git branch workflow. Add description of PR squash and merge vs merge commit. Add instructions for running only a subset of tests locally. Closes #387
closed 2021-12-02 by emiliom