echostack-org/echopype #397: Document dev workflow and testing strategy

leewujung, 2021-07-18:
We had some discussion in #378 on the development workflow, especially with the consideration of being able to update the documentation separately from the code releases (achieved via the `stable` branch that can be automatically taken by RTD). 

It also has happened a few times that our current conditional testing strategy (ie. only run a subset of tests when the associated code is changed) had missed some important, integration type of tests. I seemed to remember that @lsetiawan had put in a commit tag that could trigger _all_ tests, but I had forgotten where that was...

It'll be a good idea for us to put together a brief document to describe our code dev and documentation workflow as well as testing strategy. Where should this doc live? in the documentation under [`Contributing to echopype`](https://echopype.readthedocs.io/en/stable/contributing.html)?

Tagging @emiliom @lsetiawan @imranmaj

emiliom, 2021-07-18:
Yeah, "Contributing to echopype" seems to be the best place. There are already elements of workflow and testing-tag descriptions in there, though what's there is already outdated. But if the page gets too long, we should consider splitting off some content into a new page, so that this page remains reasonably friendly to new contributors.

FYI, I created an issue (#387) a month ago to compile updates needed for the "Contributing to echopype" page. Looks it's time to get back to those updates.

leewujung, 2021-07-18:
Ah, I had missed #387. Great!

Let's discuss this as part of the dev/release plan this Tuesday then -- the timing works out great. :)

emiliom, 2021-07-20:
For a hotfix workflow, see the Hotfix Branches section at https://www.atlassian.com/git/tutorials/comparing-workflows/gitflow-workflow

emiliom, 2021-08-17:
@lsetiawan could you describe here the conditional testing strategy that's in place? To quote @leewujung:

> It also has happened a few times that our current conditional testing strategy (ie. only run a subset of tests when the associated code is changed) had missed some important, integration type of [...]

I'll then add that to the Contributing page, after proper massaging. Thanks!

leewujung, 2021-09-27:
@emiliom @lsetiawan : checking in to see if we want this and #387 to be part of the upcoming v0.5.4 release?

emiliom, 2021-09-27:
These documentation updates don't need to be associated with releases, strictly speaking. On the other hand, targeting a release can be helpful, to add self-imposed attention and a deadline. So, yeah! But most of the work is on @lsetiawan :wink:

leewujung, 2021-09-27:
Looking at #387 again, @lsetiawan do you think you can throw in a  few sentences in there for the currently items (the last 2), so that we can close this and that issues? It would be really helpful for me and @imranmaj making PRs for sure -- I haven't been able to remember which tag and label to use for what intended behaviors. 😅   These items in theory shouldn't be affected with the pytest docker change you want to do, so it would be conducive work. Thanks!

lsetiawan, 2021-09-27:
Sorry for the delay on my part for this stuff. But I think it would be great to include these info in the next release.

leewujung, 2021-09-27:
@lsetiawan : is the next release you refer to here the upcoming v0.5.4?

lsetiawan, 2021-09-27:
Yes indeed. 🙂