echostack-org/echopype #415: Combine echodata test for EK60 ping time uses too much memory

lsetiawan, 2021-08-12:
Seems like the EK60 ping time reversal test in combine echodata uses about 8GB of memory in total when trying to combine 2, ~250MB files. The Github limit for github workflow machines is 7GB. This caused the tests to get killed early (https://github.com/OSOceanAcoustics/echopype/runs/3315893568#step:12:193)

https://github.com/OSOceanAcoustics/echopype/pull/383/files#diff-0fa7822fe165824463f0094d64a29f9e02b4ed31833647da544621fc3ed3d3b2R54-R60

@imranmaj will investigate this issue.

leewujung, 2021-08-13:
Thanks @imranmaj for investigating the memory issue! This would really pave the way for figuring out the merging process when we go to much larger files for #407.

For the combine EchoData functionality though, perhaps we can use smaller 25 MB files from the hake survey as test data? @imranmaj are there reasons why you picked these files? Like because there's time reversal that gets changes in the merging process?

imranmaj, 2021-08-13:
Yes, these specific files have the time reversal issue.

leewujung, 2021-08-13:
Ok. Are the time reversals in multiple places, or just in ping_time or location_time? I should be able to find a couple hake survey with similar issues but smaller.

imranmaj, 2021-08-13:
ping_time for sure has a reversal, I believe location_time and mru_time do not but I can check for sure in a little while.

leewujung, 2021-08-13:
@imranmaj : this file `s3://ncei-wcsd-archive/data/raw/Bell_M._Shimada/SH1707/EK60/Summer2017-D20170719-T205415.raw` has reversed `location_time`, but `ping_time` appears to be monotonically increasing.

PR #422 (2021-08-16): Use smaller files in combine_echodata tests
Closes #415
closed 2021-08-16 by imranmaj